gsilano/BebopS #3: Hello sir. I got this issue when running the command: rosdep install --from-paths src -i

geforcexfx, 2019-08-06:


geforcexfx, 2019-08-06:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bebop_simulator: Cannot locate rosdep definition for [bebop_driver]

geforcexfx, 2019-08-06:
The other steps are fine. but when getting this issue then I dont have /gazebo/command/motor_speed topic. Here are all the topic after I run roslaunch : 
rostopic echo /
/bebop/camera1/camera_info
/bebop/camera1/image_raw
/bebop/camera1/image_raw/compressed
/bebop/camera1/image_raw/compressedDepth
/bebop/camera1/image_raw/compressedDepth/parameter_descriptions
/bebop/camera1/image_raw/compressedDepth/parameter_updates
/bebop/camera1/image_raw/compressed/parameter_descriptions
/bebop/camera1/image_raw/compressed/parameter_updates
/bebop/camera1/image_raw/theora
/bebop/camera1/image_raw/theora/parameter_descriptions
/bebop/camera1/image_raw/theora/parameter_updates
/bebop/camera1/parameter_descriptions
/bebop/camera1/parameter_updates
/clock
/gazebo_gui/parameter_descriptions
/gazebo_gui/parameter_updates
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg

gsilano, 2019-08-09:
Hi @geforcexfx , thanks for opening an issue. and for finding BebopS interesting for your research activity.  Could you be more specific about the error you received and also about the ROS distro and the OS you are using?

geforcexfx, 2019-08-09:
Hi @gsilano , Thanks for your reply. My OS is ubuntu 16.04 it is freshly installed, I followed the steps from your bebops github to install BebopS package. They are all good until this step: rosdep install --from-paths src -i. Then it showed the error I mentioned above. I was stuck and I didnt know what to do, I just continued to install and it was ok again. Right now I can run roslaunch bebop_simulator bebop_without_controller.launch 
Gazebo is opened with parrot model, then if I try to publish the topic  to fly: rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'
It does nothing. Here are all the topics after I enter the command rostopic echo : 
rostopic echo /
/bebop/camera1/camera_info
/bebop/camera1/image_raw
/bebop/camera1/image_raw/compressed
/bebop/camera1/image_raw/compressedDepth
/bebop/camera1/image_raw/compressedDepth/parameter_descriptions
/bebop/camera1/image_raw/compressedDepth/parameter_updates
/bebop/camera1/image_raw/compressed/parameter_descriptions
/bebop/camera1/image_raw/compressed/parameter_updates
/bebop/camera1/image_raw/theora
/bebop/camera1/image_raw/theora/parameter_descriptions
/bebop/camera1/image_raw/theora/parameter_updates
/bebop/camera1/parameter_descriptions
/bebop/camera1/parameter_updates
/clock
/gazebo_gui/parameter_descriptions
/gazebo_gui/parameter_updates
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/rosout
/rosout_agg

gsilano, 2019-08-09:
It seems very strange to me because the only dependencies are with the `bebop_autonomy` ROS package. It is used to interface the Parrot Bebop 2 with the ROS environment. Let me uninstall and reinstall the package. Maybe, later, I will able to tell you something more.

geforcexfx, 2019-08-09:
Hello @gsilano . I just installed it from another computer. It shows the same error. Here are the printouts of steps from the bebopS instructions until the step I got the error: (the error is at the end of the printouts) https://pastebin.com/zt7bM47n . So maybe you can findout where and why. Hope it helps. Thanks

gsilano, 2019-08-10:
I saw your log file and repeated all steps necessary to install BebopS on my laptop from scratch. First of all, I didn't find any error in installing the ROS packages. Maybe, something goes wrong in the copy and paste procedure. To avoid issues, I modified the README.md file in the repository by moving those commands to two different lines. Also, I did the same with the `git checkout` commands. In particular, by changing the lines:

`cd ~/catkin_ws/src/rotors_simulator & git checkout med18`
`cd ~/catkin_ws/src/mav_comm & git checkout med18`

in 

```
$ cd ~/catkin_ws/src/rotors_simulator 
$ git checkout med18
$ cd ~/catkin_ws/src/mav_comm
$ git checkout med18

```
After the installation procedure, I run the Bebop simulation without the controller algorithm in the loop and I was able to control the propellers speed using the command `rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'`. 

Please, try again with the new commands. If you see errors, I'll try to understand what are the available solutions for fixing it together.

geforcexfx, 2019-08-10:
Thanks , I retried from $ cd ~/catkin_ws/src, it still have the same error but it could fly with pub command now, here is the log of the installation: 
https://pastebin.com/yDCVj1gm
Right now it can fly with $ rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'
and 
roslaunch bebop_simulator task1_world.launch
but when I execute this command : roslaunch bebop_simulator task2_world.launch
Then there are still some missing topics: 
ResourceNotFound: BebopS
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/cody/catkin_ws/src/mav_comm/mav_comm
ROS path [2]=/home/cody/catkin_ws/src/mav_comm/mav_msgs
ROS path [3]=/home/cody/catkin_ws/src/mav_comm/planning_msgs
ROS path [4]=/home/cody/catkin_ws/src/rotors_simulator/rotors_comm
ROS path [5]=/home/cody/catkin_ws/src/rotors_simulator/rotors_control
ROS path [6]=/home/cody/catkin_ws/src/rotors_simulator/rotors_description
ROS path [7]=/home/cody/catkin_ws/src/rotors_simulator/rotors_evaluation
ROS path [8]=/home/cody/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins
ROS path [9]=/home/cody/catkin_ws/src/BebopS
ROS path [10]=/home/cody/catkin_ws/src/rotors_simulator/rotors_gazebo
ROS path [11]=/home/cody/catkin_ws/src/rotors_simulator/rotors_joy_interface
ROS path [12]=/home/cody/catkin_ws/src/rotors_simulator/rotors_simulator
ROS path [13]=/opt/ros/kinetic/share
Here is the log of this error: https://pastebin.com/85k9zi2a
Thank you very much for your help.

geforcexfx, 2019-08-11:
Hello @gsilano . Today I remove 2 lines: 
+ "<build_depend>bebop_driver</build_depend>"
+ "<run_depend>bebop_driver</run_depend>"
 from package.xml file in BebopS folder and this is the printout: 
rosdep install --from-paths src -i
#All required rosdeps installed successfully
And when I put the 2 lines back then it showed the error again. Is there something I can add this bebop_driver dependency manually into the package? Thanks.

gsilano, 2019-08-12:
Hi @geforcexfx. I found the issue: I missed to insert, in the list of packages needed to run the simulation, the `bebop_autonomy` ROS package. To be precise, it is not necessary, but if you want to run the controller algorithm (currently not yet tested) with the real Parrot Bebop, the `bebop_autonomy` package is needed. I will modify the REAMD.md file to address the issue.

For the second, I think you have to source the bashrc file again. Please, try to run the commands reported below:

```
$ echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
```

geforcexfx, 2019-08-12:
Hello @gsilano . Thanks for your help. I think it is almost there but not there yet. I tried the commands:
 $ git clone https://github.com/AutonomyLab/bebop_autonomy.git 
$ rosdep update
$ cd ~/catkin_ws
$ rosdep install --from-paths src -i
Then this is the package it missed: 
E: Unable to locate package ros-kinetic-parrot-arsdk
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-kinetic-parrot-arsdk] failed
I googled and I found out I need to do: 
$ git clone https://github.com/AutonomyLab/parrot_arsdk.git src/parrot_arsdk
then I rosdep update again and tried rosdep install 1 more time. I still missing 1 package and I also googled it but there's no luck. Could you please help me? Thanks. This is the last log:
$rosdep install --from-paths src -i
executing command [sudo -H apt-get install ros-kinetic-joy-teleop]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-joy-teleop
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-kinetic-joy-teleop] failed

gsilano, 2019-08-12:
It seems very strange to me because I ran the same commands this morning, before answering you on GitHub, and I didn't have any issue. Perhaps, something went wrong during the installation of ROS. I suggest you follow the procedure explained on the ROS website again. I reported the link below for your convenience.

http://wiki.ros.org/kinetic/Installation/Ubuntu

geforcexfx, 2019-08-12:
Thank you very much . it helps ,this is what I did:
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
$ sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
$ sudo apt-get update
$ sudo apt-get install ros-kinetic-desktop-full #doesnt do anything, just in case something missing
$ sudo apt-get install ros-kinetic-joy
$ cd ~/catkin_ws
$ rosdep update
$ rosdep install --from-paths src -i
#All required rosdeps installed successfully
$ catkin build
$ echo "source ~/catkin_ws/devel/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
I also installed in another computer and no error. But do you know why this command:
$ roslaunch bebop_simulator task2_world.launch
Still gives the same error as some topics are missing? This is the log for task2_world.launch : 
https://pastebin.com/8BV5Rzc4

geforcexfx, 2019-08-12:
this error for task2_world.launch happens in both computer(the second one followed exactly as from readme.md no extra step and it worked)

geforcexfx, 2019-08-12:
I checked task2_world.launch file. and this line shows the error

   <group ns="$(arg name)">
       <node name="waypoint_example" pkg="bebop_simulator" type="waypoint_example" output="screen" args="$(find bebopS)/resource/waypoints.txt"/>   
    </group>
Even though I have all the required resources. Do you know why it doesn't work? Thanks

gsilano, 2019-08-13:
I just fixed the typo. Thanks for finding it. Please, try now and let me know.

geforcexfx, 2019-08-13:
Thanks for telling me. $(find BebopS) doesn't work either. I tried to change to all $(find BebopS) commands by my local directory path into these lines then it works:
<rosparam command="load" file="/home/cody/catkin_ws/src/BebopS/resource/controller_bebop.yaml" />
      <rosparam command="load" file="/home/cody/catkin_ws/src/BebopS/resource/bebop.yaml" />
      <rosparam command="load" file="/home/cody/catkin_ws/src/BebopS/resource/EKF_matrix.yaml" />
      <rosparam command="load" file="/home/cody/catkin_ws/src/BebopS/resource/waypoint_filter.yaml" />
 <node name="waypoint_example" pkg="bebop_simulator" type="waypoint_example" output="screen" args="/home/cody/catkin_ws/src/BebopS/resource/waypoints.txt"/>   
    </group>

gsilano, 2019-08-13:
It seems very strange to me. It worked when I tested. Okay, let's try another way. Please, consider the latest commit.

geforcexfx, 2019-08-16:
Hello @gsilano  . Thanks for your help. Now task1_world.launch and task2_world.launch work after pulling from the newest commit. I forgot to tell you that this command haven't worked since the first day I tried:
roslaunch bebop_simulator bebop_without_controller.launch enable_odometry_sensor_with_noise:=true
Could you please also take a look at it? Thanks. Here is the log of it : https://pastebin.com/u0MWggAj

gsilano, 2019-08-16:
I think I have fixed the bug with my last commit. Please, let me know.

If all issues have been solved, please do not forget to close the issue.

geforcexfx, 2019-08-16:
Thanks. All issues are fixed.
Best regards

closed 2019-08-16 by geforcexfx

Raviteja-T, 2020-12-11:
root@7fe7b2744d5d:~/carla-ros-bridge/catkin_ws# sudo rosdep init
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.

Raviteja-T, 2020-12-11:
[image]