gsilano/BebopS #4: No motor spinning

jocacace, 2019-10-26:
Hi, 
I am trying to use the bebop simulator on my system based on ROS melodic (and hence Gazebo 9). The compilation works well but when I launch the task2_world.launch file the working scene is correctly loaded (the bebop uav is land) but the mission doesn't start. The rotors of the robot are completely stop. 

I am wandering if Gazebo 9 is supported... No errors appear on the terminal. Do you have some suggestions to debug this problem?

Regards,

gsilano, 2019-10-28:
I'm investigating the issue, and I will give you back ASAP.

jocacace, 2019-10-28:
Dear @gsilano 

Actually I found a problem with my setup. I have not switched on correct banch for rotors_simulator... so I think I am using different plugins to spawn the bebop robot...

gsilano, 2019-10-28:
Nice. So, can I close the issue? Everything is fine?

jocacace, 2019-10-28:
Yes, it works :)!

Thank you

closed 2019-10-28 by jocacace