gsilano/BebopS #9: Problem with "measurement_delay"

valarsundarlis, 2020-01-04:
Hey @gsilano, after adding the changes you mentioned in #8 I got the following error:

valarsundarlis@deadspace:~$ roslaunch bebop_simulator bebop_without_controller.launch
... logging to /home/valarsundarlis/.ros/log/3e23b4da-2edb-11ea-abd3-84ef187d1423/roslaunch-deadspace-4526.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Invalid parameter "measurement_delay" None None
when instantiating macro: imu_plugin_macro (/home/valarsundarlis/ROS/catkin_ws/src/rotors_simulator/rotors_description/urdf/component_snippets.xacro)
in file: /home/valarsundarlis/ROS/catkin_ws/src/BebopS/urdf/bebop.urdf.xacro
while processing /home/valarsundarlis/ROS/catkin_ws/src/BebopS/launch/spawn_bebop.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [     /opt/ros/kinetic/lib/xacro/xacro --inorder '/home/valarsundarlis/ROS/catkin_ws/src/BebopS/urdf/bebop.urdf.xacro'     enable_ground_truth:='true'     enable_odometry_sensor_with_noise:='false'     disable_odometry_sensor_with_noise:='true'     enable_ground_truth_sensor:='false'     enable_wind_plugin:='false'     enable_laser1D:='false'     enable_imu:='true'     namespace:=bebop     wind_force:=0.25     wind_start:=5.0     wind_duration:=40.0     wind_direction_x:=1.0     wind_direction_y:=0.0     wind_direction_z:=0.0     ] returned with code [2]. 

Param xml is <param command="     $(find xacro)/xacro --inorder '$(arg model)'     enable_ground_truth:='$(arg enable_ground_truth)'     enable_odometry_sensor_with_noise:='$(arg enable_odometry_sensor_with_noise)'     disable_odometry_sensor_with_noise:='$(arg disable_odometry_sensor_with_noise)'     enable_ground_truth_sensor:='$(arg enable_ground_truth_sensor)'     enable_wind_plugin:='$(arg enable_wind_plugin)'     enable_laser1D:='$(arg enable_laser1D)'     enable_imu:='$(arg enable_imu)'     namespace:=$(arg name)     wind_force:=$(arg wind_force)     wind_start:=$(arg wind_start)     wind_duration:=$(arg wind_duration)     wind_direction_x:=$(arg wind_direction_x)     wind_direction_y:=$(arg wind_direction_y)     wind_direction_z:=$(arg wind_direction_z)     " name="robot_description"/>
The traceback for the exception was written to the log file

gsilano, 2020-01-04:
The commit should fix the error. Please, let me know if other changes are needed. If not, I will close the issue

valarsundarlis, 2020-01-04:
The command now runs successfully and gazebo opened. Thank you so much!

closed 2020-01-04 by gsilano