gsilano/BebopS #10: About "message type: mav_msgs/Actuators"

chanforg, 2020-03-01:
I have installed BebopS on my Ubuntu 16.04 with ROS kinetic and Gazebo 7. 
I run  the command 'roslaunch bebop_simulator bebop_without_controller.launch' successfully, but when i typed the command "rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}" 
it told me  :
 ERROR: invalid message type: mav_msgs/Actuators.
If this is a valid message type, perhaps you need to type 'rosmake mav_msgs'
Then i type "rosmsg list" to check, my computer really lack the the message type.
so i type 'rosmake mav_msgs', but it didn't work.
Thank you for any help :)

gsilano, 2020-03-01:
Hi @chanforg! First of all, thank you for finding BebopS useful to you. As for your question, could it be a wrong copy and paste?
I mean, seeing your message a ` (at the end of the command) is missing. The message should be:

```
rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'
```

And, if it does not work. Please, try with the one suggested below (I reported it for your convenience)

```
rostopic pub /command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'
```

chanforg, 2020-03-01:
[image]
It still not work

gsilano, 2020-03-01:
So, as you wrote, the 'roslaunch bebop_simulator bebop_without_controller.launch' command works. So, it should not be a problem related to the `mav_comm` package.

To understand what is the problem, I think it would be interesting to understand what happens when you run the 'roslaunch bebop_simulator bebop_without_controller.launch' command. In particular, when the simulation is running, in another terminal window run the `rosnode info position_controller_node` and `rostopic info NAME_OF_THE_TOPIC` (where the propellers angular velocities are published) and paste the screens here.

chanforg, 2020-03-01:
[image]
[image]

gsilano, 2020-03-01:
No, sorry. I think there is something wrong. Did you run the command when the launch file is running?

chanforg, 2020-03-01:
> No, sorry. I think there is something wrong. Did you run the command when the launch file is running?

I am  sure I have run the launch
[image]

gsilano, 2020-03-01:
And there is something that is wrong. It is very strange that, if the simulation is running, any nodes are displayed on the terminal line.

Could you copy and paste the verbose you see when running the simulation?

chanforg, 2020-03-02:
`$ roslaunch bebop_simulator bebop_without_controller.launch
`... logging to /home/ethan/.ros/log/b687e82e-5c30-11ea-9f19-98e7f44cf5fc/roslaunch-ethan-HP-Pavilion-Notebook-2877.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ethan-HP-Pavilion-Notebook:33299/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /tf_prefix: bebop
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_robot (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [2892]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b687e82e-5c30-11ea-9f19-98e7f44cf5fc
process[rosout-1]: started with pid [2905]
started core service [/rosout]
process[gazebo-2]: started with pid [2908]
process[gazebo_gui-3]: started with pid [2912]
process[spawn_robot-4]: started with pid [2918]
SpawnModel script started
[INFO] [1583117498.606854, 0.000000]: Loading model XML from ros parameter
[INFO] [1583117498.615400, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1583117499.806817224]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1583117499.810260917]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1583117499.810721513]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1583117499.812324849]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1583117501.945120, 0.000000]: Calling service /gazebo/spawn_urdf_model
[ INFO] [1583117502.156852611, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1583117502.160872374, 0.027000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1583117502.466920, 0.195000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1583117502.491282597, 0.195000000]: Physics dynamic reconfigure ready.
[ INFO] [1583117502.491902042, 0.195000000]: Physics dynamic reconfigure ready.
[spawn_robot-4] process has finished cleanly
log file: /home/ethan/.ros/log/b687e82e-5c30-11ea-9f19-98e7f44cf5fc/spawn_robot-4*.log
[ INFO] [1583117503.011917346, 0.195000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1583117503.298053254, 0.195000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to "bebop"

gsilano, 2020-03-02:
Finally, I got the mistake. Yes, you are able to run the launch file but not the simulation. I mean, the drone does not take off. You see it in Gazebo, but the controller is not in the loop (see the active nodes). I think you did not follow the instructions in the right way or a step is missing.

Please, reinstall everything paying attention to respect the exact order.

chanforg, 2020-03-02:
oh! yes i reinstall it .and now it can be work!
Thank you for your patient answer，Please forgive my carelessness, I have just touched this field。:-)

gsilano, 2020-03-02:
Take it easy, it can happen. I happy to ear from you that everything is fine now. So, I can close the issue.

closed 2020-03-02 by gsilano