gsilano/BebopS #14: How does the thresholding in the position controller work?

Luke-A-F, 2020-04-14:
Hi,
The following code is within the position controller for the x,y,z direction.

*u_z = ( (e_z_  * K_z_1_* K_z_2_)/lambda_z_ ) + ( (dot_e_z_ * K_z_2_)/lambda_z_ );
  
   if (*u_z > 1 || *u_z <-1)
	   if (*u_z > 1)
		   *u_z = 1;
	   else
		   *u_z = -1;
	  
   *u_z = (*u_z * 1/2) + ( (K_z_1_/lambda_z_) * dot_e_z_ );
   
   if (*u_z > 1 || *u_z <-1)
	   if (*u_z > 1)
		   *u_z = 1;
	   else
		   *u_z = -1;
	 *u_z = m_* ( *u_z * lambda_z_);

I am trying to modify the gain values of the position controller to reject disturbance. I also notice that there is not thresholding in the attitude controller. How does the thresholding in the position controller work?

gsilano, 2020-04-14:
Have you read the reference paper listed in the Wiki?

Luke-A-F, 2020-04-14:
I just now reread the paper. So, the thresholding provides stability. 
Is it possible to modify the control input before or after the thresholding and still keep it within the simulation bounds with the virtual input bounds?

gsilano, 2020-04-14:
What do you mean "change control input"? Which part of the control algorithm are you referring to?

Luke-A-F, 2020-04-14:
The control input in the position controller for the x,y, and z direction. 
From the original post, for example, is it possible to modify like u_z = m_ ( *u_z + modifier * lambda_z_); and still keep the simulation valid?

gsilano, 2020-04-14:
If you change `u_z`, you change the control law. Therefore, I cannot say whether the control algorithm is still stable. However, I can obviously say that it will not work for the scenario considered.

closed 2020-04-14 by Luke-A-F