gsilano/BebopS #15: Improvement : Changing Laser1D to sonar

tomlogan501, 2020-04-15:
Hi,
Great package ! i will use it for one of our student.
I notice you are using the laser plugin to do a equivalent of a sonar. Is it intentional ?
For example on hector_quad, there is a sonar and a laser scan downward or horizontal.
I can create a PR if you want.

gsilano, 2020-04-15:
Hi @tomlogan501!! As it is described in the README file, BebopS is based on the ROS package RotorS. Therefore, if a 1D laser scanner is listed as an available sensor, it comes from the RotorS package. Regarding the experiments conducted with the platform (i.e., the listed launch files), no 1D laser scanner was used.

Before creating a pull request, we should understand if these sensors are also available in RotorS and not present in BebopS because it is not aligned with the latest release. So, if not, understand if the sensors are generic or specific. I mean, if the software module that models that sensor considering a specific sensor (e.g.,  MB1340 or the MB1240 XL-MaxSonar-EZ sensors) or aims to model only the sensor behavior.

tomlogan501, 2020-04-15:
I don't see any plugin to do the job in rotors.
In the package gazebo_ros_plugins, the one that could do it, is **gazebo_ros_range**, for a generic behavior.
Well then I should update it on an independent branch on our repository or the student will load it from a customized xacro.

gsilano, 2020-04-15:
Yes, you are right. I double-checked the contents of the `component_snippets.xacro` file, in the `rotors_description/urdf` folder. There is no sonar sensor.
If you could add it, it would be awesome.

tomlogan501, 2020-04-15:
I can add it but your package will depend on **gazebo_plugins** then. Should it be a problem even if gazebo is 90% of the time installed with ROS :) ?

gsilano, 2020-04-15:
No, it shouldn't be a problem. Thank you!

gsilano, 2020-08-15:
Any update on this issue? If not, I will close it.

tomlogan501, 2020-08-19:
Sorry I didn't make the job but you can keep the idea

gsilano, 2020-08-19:
Okay, I will close the issue for now.

closed 2020-08-19 by gsilano