gsilano/BebopS #18: Where are the values from Table I and II derived from the reference paper?

Luke-A-F, 2020-04-21:
I am trying to manipulate the values from the position control law in the reference paper and possibly change the virtual bound inputs, but I do not see anywhere in the reference paper where the values for Table I and Table II are derived. When I do the math for max propeller speed from Table I with equation 5, it does not equal 18.6N that is in Table II as a physical Ut constraint.

Where are the values from Table I and II derived from that are in the reference paper?

gsilano, 2020-04-21:
As you can read from the README.md file, the software platform is an extract of the work carried out for the industrial challenge of the 26th Mediterranean Conference on Control and Automation (MED'18) in which we participated ([here the conference report](https://ieeexplore.ieee.org/document/8667511)). The values shown in Tables I and II come from the description of the challenge.

I hope this helps.

Luke-A-F, 2020-04-23:
Hm. I could not find the description of the challenge values in the conference report or the website, but thank you anyway.

gsilano, 2020-04-23:
No! The report describes the event and the challenge. I suggest you to send me an e-mail.

closed 2020-04-23 by gsilano