gsilano/BebopS #20: Problem when implementing robot swarm

sheldonjinqi, 2020-05-18:
Hi,
I am trying to use the BebopS to run some experiments on a team of robots. Right now I am having trouble with launching multiple robots and keep them at hovering. 

This is the error i'm getting when i try to roslaunch my launch file. It seems like the gazebo node is publishing two topics "motor speed" with different data types. Any suggestions on how to fix this? 

`started roslaunch server http://sheldon-Thinkpad:37461/

SUMMARY
========

PARAMETERS
 * /bebop1/position_controller_node/EKFActive: False
 * /bebop1/position_controller_node/H: 0.01
 * /bebop1/position_controller_node/Qp/aa: 0.001
 * /bebop1/position_controller_node/Qp/bb: 0.001
 * /bebop1/position_controller_node/Qp/cc: 0.001
 * /bebop1/position_controller_node/Qp/dd: 0.001
 * /bebop1/position_controller_node/Qp/ee: 0.001
 * /bebop1/position_controller_node/Qp/ff: 0.001
 * /bebop1/position_controller_node/Tsf: 1.5
 * /bebop1/position_controller_node/U_xyz/U_x: 1.181
 * /bebop1/position_controller_node/U_xyz/U_y: 1.181
 * /bebop1/position_controller_node/U_xyz/U_z: 4.6697
 * /bebop1/position_controller_node/beta_phi/beta_phi: -1.7514
 * /bebop1/position_controller_node/beta_psi/beta_psi: -14.3431
 * /bebop1/position_controller_node/beta_theta/beta_theta: -1.7513
 * /bebop1/position_controller_node/beta_xy/beta_x: -26.4259
 * /bebop1/position_controller_node/beta_xy/beta_y: -26.3627
 * /bebop1/position_controller_node/beta_z/beta_z: -27.2277
 * /bebop1/position_controller_node/bf: 8.54858e-06
 * /bebop1/position_controller_node/bm: 0.016
 * /bebop1/position_controller_node/csvFilesStoring: False
 * /bebop1/position_controller_node/csvFilesStoringTime: 15.0
 * /bebop1/position_controller_node/dev_vx: 0.01
 * /bebop1/position_controller_node/dev_vy: 0.01
 * /bebop1/position_controller_node/dev_vz: 0.01
 * /bebop1/position_controller_node/dev_x: 0.01
 * /bebop1/position_controller_node/dev_y: 0.01
 * /bebop1/position_controller_node/dev_z: 0.01
 * /bebop1/position_controller_node/inertia/xx: 0.00389
 * /bebop1/position_controller_node/inertia/xy: 0.0
 * /bebop1/position_controller_node/inertia/xz: 0.0
 * /bebop1/position_controller_node/inertia/yy: 0.00389
 * /bebop1/position_controller_node/inertia/yz: 0.0
 * /bebop1/position_controller_node/inertia/zz: 0.0078
 * /bebop1/position_controller_node/l: 0.12905
 * /bebop1/position_controller_node/mass: 0.5
 * /bebop1/position_controller_node/mu_phi/mu_phi: 0.0544
 * /bebop1/position_controller_node/mu_psi/mu_psi: 0.44
 * /bebop1/position_controller_node/mu_theta/mu_theta: 0.0543
 * /bebop1/position_controller_node/mu_xy/mu_x: 1
 * /bebop1/position_controller_node/mu_xy/mu_y: 1
 * /bebop1/position_controller_node/mu_z/mu_z: 1
 * /bebop1/position_controller_node/use_sim_time: True
 * /bebop1/position_controller_node/user_account: giuseppe
 * /bebop1/position_controller_node/waypoint_filter: True
 * /bebop1/robot_description: <?xml version="1....
 * /bebop1/tf_prefix: bebop
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
  /bebop1/
    position_controller_node (bebop_simulator/position_controller_node)
    spawn_robot (gazebo_ros/spawn_model)
[... 103 lines cut ...]
       <arg name="enable_imu" value="$(arg enable_imu)"/>
    </include>

    
    <node name="position_controller_node" pkg="bebop_simulator" type="position_controller_node" output="screen"> 
      <rosparam command="load" file="$(find bebop_simulator)/resource/controller_bebop.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/bebop.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/EKF_matrix.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/waypoint_filter.yaml" />
      <param name="use_sim_time" value="$(arg use_sim_time)" />
      <param name="csvFilesStoring" value="$(arg csvFilesStoring)"/>
      <param name="csvFilesStoringTime" value="$(arg csvFilesStoringTime)"/>
      <param name="user_account" value="$(arg user_account)"/>
      <param name="waypoint_filter" value="$(arg waypoint_filter)"/>
      <param name="EKFActive" value="$(arg EKFActive)"/>
      
      <remap from="/odometry" to="/bebop1/odometry" />
      <remap from="/odometry_gt" to="/bebop/odometry_gt" />
      <remap from="/referenceAngles" to="/bebop/referenceAngles" />
      <remap from="/filteredOutput" to="/bebop/filteredOutput" />
      <remap from="/stateErrors" to="/bebop/stateErrors" />
      <remap from="/smoothedTrajectory" to="/bebop/smoothedTrajectory" />
      <remap from="/command/trajectory" to="/$(arg mav_name)1/command/trajectory" />
    </node>
    </group>`

gsilano, 2020-05-20:
Hi @sheldonjinqi! Have you tried to make a launch file as the one [at this link](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/crazyflie2_swarm_hovering_example.launch)?

sheldonjinqi, 2020-05-20:
Hi @gsilano, I used crazyflie2_swarm_hovering_example.launch as a reference when I wrote my launch file. I used the group tag for each robot with a specific namespace. I believe i have found the problem. Without the group tag, there are two topics when running the launch file, /motor_speed and /bebop1/motor_speed. After putting everything in the group tag with namespace bebop_1, the first topic would also have a name /bebop_1/motor_speed and leads to the name conflict.

gsilano, 2020-05-20:
Ok, awesome. If you have discovered the issue and now everything works. Could you provide the swarm's example through a PR?

sheldonjinqi, 2020-05-20:
Definitely, I would do it later. Another general question regarding the controller. With the provided position controller, when running task2_world, the robot moves slowly and stops at each waypoint. Any suggestion for designing a smoother trajectory such that the robot flies continuously without stopping at each waypoint. I am pretty new to this field, so any suggestion would be great. 
Thanks in advance for your help :)

gsilano, 2020-05-20:
A possibility could be to replace the waypoint filter (for more details see [the reference paper](https://github.com/gsilano/BebopS/wiki/Publications)) with a [spline trajectory generator](https://github.com/gsilano/BebopS/blob/feature/spline/src/nodes/hovering_example_spline.cpp). I've already started developing it, but I don't have time to finalize the code. You could start with this.

sheldonjinqi, 2020-05-20:
thanks! i will take a look at the paper and try it myself.

PR #29 (2020-08-26): Enable compatibility with multiple bebops
Modifications made to [urdf/multirotor_base.urdf.xacro](https://github.com/Collin-Thornton/BebopS/blob/multi_bebop_devel/urdf/multirotor_base.urdf.xacro) such that multiple bebops may be spawned
in the same simulation. Lines 109 and 113 were changed so that parameter titles matched with those
found in [RotorS/rotors_gazebo_plugins/src/gazebo_motor_model.cpp](https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_gazebo_plugins/src/gazebo_motor_model.cpp). This resolves the rostopic
advertising conflict on bebop/motor_speed experienced by other [... cut]
commit d5fb85ff (2020-08-26): Enable compatibility with multiple bebops

Modifications made to urdf/multirotor_base.urdf.xacro such that multiple bebops may be spawned
in the same simulation. Lines 109 and 113 were changed so that parameter titles matched with those
found in RotorS/rotors_gazebo_plugins/src/gazebo_motor_model.cpp. This resolves the rostopic
advertising conflict on bebop/motor_speed experienced by other developers.

Launch file examples have been added. See

    - launch/multiple_bebop_without_controller.launch

    - launch/multiple_task2_world.launch

for examples without and with a controller, respective [... cut]
closed 2020-08-26 by gsilano