gsilano/BebopS #21: how to control the position of the propeller

octopusF, 2020-06-21:
hi, I want to control the position of the Bebop's  propeller, but I do not know how to do it. In the multirotor_base. urdf. xacro, I notice the plugin "librotors_gazebo_motor_model.so" , and I think it controls the velocity of the propeller  ,  I want to know if the plugin "librotors_gazebo_motor_model.so" could control the position of the propeller?  And how? 
Thank you.

gsilano, 2020-06-21:
What do you mean with "the position of the propeller"? If you want to change the geometry, i.e., the distance of the propellers in terms of x and y-coordinates, just change this line

https://github.com/gsilano/BebopS/blob/79d4c8d12dea0819aa7a63556a73e4f9a97aac8d/urdf/bebop_base.urdf.xacro#L72-L73

https://github.com/gsilano/BebopS/blob/79d4c8d12dea0819aa7a63556a73e4f9a97aac8d/urdf/bebop_base.urdf.xacro#L95-L96

https://github.com/gsilano/BebopS/blob/79d4c8d12dea0819aa7a63556a73e4f9a97aac8d/urdf/bebop_base.urdf.xacro#L120-L121

https://github.com/gsilano/BebopS/blob/79d4c8d12dea0819aa7a63556a73e4f9a97aac8d/urdf/bebop_base.urdf.xacro#L145-L146

octopusF, 2020-06-21:
hi,@gsilano. I did not make myself clear. The velocity of the propeller can be controlled by sending commands in the terminal:
$ rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'
I sent the command
 $ rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angles: [1, 1, 1, 1]}'
to control the angle of the propeller, but this does not work. So how to change the angle of the propeller. 
Thank you.

gsilano, 2020-06-21:
If you need to change it in real-time, during the simulation, as I understand, you would like to simulate a fully actuated platform as described in these papers. 

[Comparison of Control Methods for Trajectory Tracking in Fully Actuated Unmanned Aerial Vehicles](https://ieeexplore.ieee.org/document/9097927)
[Full Pose Tracking for a Tilt-Arm Quadrotor UAV](https://ieeexplore.ieee.org/document/8511566)
[Modeling and control of FAST-Hex: A fully-actuated by synchronized-tilting hexarotor](https://ieeexplore.ieee.org/document/7759271)

Instead, if you need to change the rotors configuration before starting the simulation, simply change the `rpy="0 0 0` of the part of the code mentioned before.

octopusF, 2020-06-21:
@gsilano thank you for your reply. I think I have got the answer.

closed 2020-06-22 by gsilano