gsilano/BebopS #22: Where can I find the control algorithm file

octopusF, 2020-06-30:
Hi, I want to use bebops to verify my control algorithm, where can I find the control algorithm file .
thank you.

gsilano, 2020-06-30:
I don't know if I got well your question. However, in the [Wiki](https://github.com/gsilano/BebopS/wiki/Publications) is the link to the paper linked to this work. As for the controller, it is available in the `src` folder of the repo.

gsilano, 2020-11-01:
Hi @octopusF! Can I close this issue?

JamalShams, 2020-11-19:
Hi @gsilano I wanna test my algorithm that is written in .py 
How can i replace the algorithm to check mine for the same drone?

gsilano, 2020-11-19:
What do you would like to test? The controller?

JamalShams, 2020-11-20:
Hi @gsilano I wanna test my algorithm that is written in python. 
How can i replace the algorithm to check mine for the same drone?

JamalShams, 2020-11-20:
> What would you like to test? The controller?

i want to test my algorithm. 
Actually i wanna replace my control algorithm with yours, just to ensure that i am working on a right track

JamalShams, 2020-11-20:
is it `controller_bebop.yaml`  ?

gsilano, 2020-11-20:
Okay, I get it. Here are the steps to make it work with the simulator:

1. Put your controller in the `src\library` folder.
2. Work with a copy or modify the `position_controller_with_bebop_node.cpp` file, and the relative heading file (`position_controller_with_bebop_node.h`), to work with the new controller algorithm (your controller).
3. Edit the `CMakeLists.txt` file to work with the new setup. I don't think it is necessary to change anything since your code is written in Python, but it would be better to check if it fits with your controller.

JamalShams, 2020-11-23:
Thank you for your reply i try it and will reply you.

JamalShams, 2020-12-21:
Isn't it `position_controller_with_bebop.cpp` rather than  `position_controller_with_bebop_node.cpp`? because in the `src\library` i found the former mentioned file.

gsilano, 2020-12-21:
Yes, you are right. My mistake

JamalShams, 2020-12-21:
Sorry, i found the node. that is not in `src\library` rather it is in `src\nodes`

gsilano, 2020-12-21:
I updated my previous message and deleted others to improve the readability of the issue. The controller algorithms are in the `libray` folder, while the node (it can use various control algorithms, look at [this repository](https://github.com/gsilano/CrazyS) to understand what I mean) is in the `nodes` folder. You can understand more about the software architecture by reading [these publications](https://github.com/gsilano/CrazyS/wiki/Publications).
I hope this helps.

JamalShams, 2020-12-22:
Sorry to bother you so much @gsilano . but don't you think the above discussion was about  BebopS rather than CrazyS.
I am saying this because [this repository](https://github.com/gsilano/CrazyS) leads to the discussion about CrazyS. I am sorry but please guide me if i am wrong in this regard.

gsilano, 2020-12-22:
Ok, let's start from the beginning. Your question was: how I can insert my control algorithm in BebopS? It is written in Python.
To use your control algorithm, you have to create an interface with the existing nodes. At this time, the nodes use the scripts contained in the `library` folder. To help you understand how to connect various control algorithms with a single node, I suggested you to take a look at CrazyS. If you see the repository, especially the [`position_controller.cpp` file](https://github.com/gsilano/CrazyS/blob/master/rotors_control/src/nodes/position_controller_node.cpp) you will find out how various control algorithms can run together. Then, it is just a matter of connecting yours. 
Before running the script, you need to update the `CMakeList.txt` file. 

I hope this helps.

closed 2021-06-26 by stale[bot]