gsilano/BebopS #23: How can I takeoff the drone in gazebo through the /command/motor_speed topic

danielrs975, 2020-07-15:
Hello

I have a question, how can I takeoff the drone using the motor_speed topic of ROS?. I am been trying but without success

gsilano, 2020-07-16:
Hi @danielrs975! It seems strange that you are not able to publish on the `/command/motor_speed` topic. However, the problem may be related to the version of Gazebo. Have you tried seeing in the issue tracker? Perhaps, the problem has already been solved by someone.

danielrs975, 2020-07-16:
Hi! I am able to publish on the topic but I do not find the right values to make the drone takeoff

gsilano, 2020-07-16:
What do you mean by"the right values"? Have you seen that propellers move when you publish values on the topic?

danielrs975, 2020-07-16:
Yes the propellers move, for example I publish this angular velocities [1000, 1000, 1000, 1000] and the drone does not take off

gsilano, 2020-07-16:
Try to increase that value. I don't remember the value that allows the drone to take off. You can see the maximum propeller speeds [in this file](https://github.com/gsilano/BebopS/blob/master/urdf/bebop_base.urdf.xacro).

danielrs975, 2020-07-16:
Thank you for the help :)

closed 2020-07-16 by gsilano