gsilano/BebopS #24: got Failed     << bebop_driver:make when install

mei123hao, 2020-07-16:
I am on ubuntu 18.04 .I got error when catkin build . The error show like the following：
cd /home/viets/worksapce/simulator/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/cmake /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/viets/worksapce/simulator/bebop_ws/devel/.private/bebop_driver -DCMAKE_INSTALL_PREFIX=/home/viets/worksapce/simulator/bebop_ws/install; cd -
...............................................................................................................................................
_______________________________________________________________________________________________________________________________________________
Errors     << bebop_driver:make /home/viets/worksapce/simulator/bebop_ws/logs/bebop_driver/build.make.000.log                                  
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function ‘bool bebop_driver::VideoDecoder::ReallocateBuffers()’:
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:32: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
     const uint32_t num_bytes = avpicture_get_size(AV_PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5653:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:32: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
     const uint32_t num_bytes = avpicture_get_size(AV_PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5653:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:136:113: warning: ‘int avpicture_get_size(AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
     const uint32_t num_bytes = avpicture_get_size(AV_PIX_FMT_RGB24, codec_ctx_ptr_->width, codec_ctx_ptr_->width);
                                                                                                                 ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5653:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:145:27: warning: ‘int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
     ThrowOnCondition(0 == avpicture_fill(
                           ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5638:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:145:27: warning: ‘int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
     ThrowOnCondition(0 == avpicture_fill(
                           ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5638:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:147:69: warning: ‘int avpicture_fill(AVPicture*, const uint8_t*, AVPixelFormat, int, int)’ is deprecated [-Wdeprecated-declarations]
                        codec_ctx_ptr_->width, codec_ctx_ptr_->height),
                                                                     ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:5638:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp: In member function ‘bool bebop_driver::VideoDecoder::Decode(const ARCONTROLLER_Frame_t*)’:
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:267:25: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                         ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4993:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
     ^
/home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:267:25: warning: ‘int avcodec_decode_video2(AVCodecContext*, AVFrame*, int*, const AVPacket*)’ is deprecated [-Wdeprecated-declarations]
     const int32_t len = avcodec_decode_video2(codec_ctx_ptr_, frame_ptr_, &frame_finished, &packet_);
                         ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/include/bebop_driver/bebop_video_decoder.h:32:0,
                 from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_video_decoder.cpp:25:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4993:5: note: declared here
 int avcodec_decode_video2(AVCodecContext *avctx, AVFrame *picture,
[... 484 lines cut ...]
           ^
In file included from /home/viets/worksapce/simulator/bebop_ws/src/bebop_autonomy/bebop_driver/src/bebop_driver_nodelet.cpp:31:0:
/opt/ros/melodic/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1028:26: error: expected primary-expression before ‘,’ token
   std::array<tf2::Vector3, 2> tmp_array;
                          ^
/opt/ros/melodic/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1028:31: error: ‘tmp_array’ was not declared in this scope
   std::array<tf2::Vector3, 2> tmp_array;
                               ^
make[2]: *** [CMakeFiles/bebop_driver_nodelet.dir/src/bebop_driver_nodelet.cpp.o] Error 1
make[1]: *** [CMakeFiles/bebop_driver_nodelet.dir/all] Error 2
make: *** [all] Error 2
cd /home/viets/worksapce/simulator/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
[... cut ...]
[build]   Abandoned: 1 packages were abandoned.                                                                                                
[build]   Failed:    1 packages failed.                                                                                                        
[build] Runtime: 28.3 seconds total.                                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

Anyone know how to  solve this problem？Help！!

gsilano, 2020-07-16:
Have you followed all the steps reported in the README file?

mei123hao, 2020-07-16:
> Have you followed all the steps reported in the README file?

Thanks for your replying！ I am sure that I follow all the  instructions for ubuntu18.04,  I delete the catkin workspace and try again, but I get the same error：
Failed     << bebop_driver:make                     [ Exited with code 2 ]                                                                     
Failed    <<< bebop_driver                          [ 18.2 seconds ]                                                                           
Abandoned <<< bebop_tools                           [ Unrelated job failed ]                                                                   
Abandoned <<< rotors_gazebo_plugins                 [ Unrelated job failed ]                                                                   
Abandoned <<< bebop_simulator                       [ Unrelated job failed ]                                                                   
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]                                                                   
Finished  <<< rotors_control                        [ 18.7 seconds ]                                                                           
[build] Summary: 12 of 17 packages succeeded.                                                                                                  
[build]   Ignored:   3 packages were skipped or are blacklisted.                                                                               
[build]   Warnings:  2 packages succeeded with warnings.                                                                                       
[build]   Abandoned: 4 packages were abandoned.                                                                                                
[build]   Failed:    1 packages failed.                                                                                                        
[build] Runtime: 31.5 seconds total.                                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

gsilano, 2020-07-16:
Have you run the command `rosdep install --from-paths src -i` in the `catkin_ws` folder before `catkin build`?

mei123hao, 2020-07-17:
> Have you run the command `rosdep install --from-paths src -i` in the `catkin_ws` folder before `catkin build`?

I am sure I have run the command `rosdep install --from-paths src -i`.
[image]
but I still get error :
[image]
[image]
[image]

gsilano, 2020-07-17:
This seems very strange to me. Do you confirm that you have also installed the following packages?

```
$ git clone -b med18 https://github.com/gsilano/rotors_simulator.git
$ git clone -b med18 https://github.com/gsilano/mav_comm.git
$ git clone https://github.com/gsilano/BebopS.git
$ git clone https://github.com/AutonomyLab/bebop_autonomy.git
```

Could you send me a photo of the `src` folder?

mei123hao, 2020-07-17:
This is the photo of src floder:
[image]
Is this the problem of the bebop_autonomy packages? I tried to make a standlone workspace for bebop_autonomy and catkin build it. It reports the similar error.

gsilano, 2020-07-17:
It may be. Have you tried to compile only the `bebop_autonomy` package? Perhaps, the problem is a missing dependence and is not closely related to BebopS.

closed 2020-11-24 by stale[bot]