gsilano/BebopS #25: UAV can't move

golden-memory, 2020-07-29:
excuse me, if i make some mistakes
[image]

gsilano, 2020-07-29:
This seems to be an old problem. Have you tried to see if there are similar problems like yours in the issue tracker? However, first of all, I would see if everything works by executing a few simple commands, e.g., `rqt` to have a node graph and `rostopic list` to see the list of available topics. Then, give me back with the results and I will try to help you in some way.

golden-memory, 2020-07-29:
> This seems to be an old problem. Have you tried to see if there are similar problems like yours in the issue tracker? However, first of all, I would see if everything works by executing a few simple [...]

Thank a lot! My installation environment is Ubuntu 16.04 with ROS Kinetic and Gazebo 7
When executing `roslaunch bebop_simulator bebop_without_controller.launch`, it shows that
[image]
[image]
--------------------------------------------------------------------------------------------------------------------------------------------------------
when executing `rqt_graph`, it shows that
[image]
[image]
[image]
--------------------------------------------------------------------------------------------------------------------------------------------------------
when executing `rostopic list`, it shows that
[image]
--------------------------------------------------------------------------------------------------------------------------------------------------------
my problem is, when executing `rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'` , it occurs error
[image]

gsilano, 2020-07-29:
Ye, I got it. As you can see, there is something wrong with the Gazebo and/or ROS environment. For this reason, the propeller angular velocities command cannot be sent. I suggest taking a look at previous issues.  For example, this one: https://github.com/gsilano/BebopS/issues/23

golden-memory, 2020-07-30:
> Ye, I got it. As you can see, there is something wrong with the Gazebo and/or ROS environment. For this reason, the propeller angular velocities command cannot be sent. I suggest taking a look at previous issues. For example, this one: #23

Oh, it seems a old question. Anyway, Thanks for your guidance. I will continue to pay attention to this issue. Besides, I found I can control the UAV by using the `waypoints.txt`, it's enough for me temporary.

closed 2020-07-30 by gsilano