gsilano/BebopS #26: How to use RotorS’ world model

golden-memory, 2020-07-30:
I want to use the world model from RotorS, but it shows as following (drone upside down). Could anyone tell me the reason? or if i need to modify the .world file?
[image]
[image]
[image]

gsilano, 2020-07-30:
Copy and paste the content of the launch and world file

golden-memory, 2020-07-30:
> Copy and paste the content of the launch and world file

basic1.world form RotorS’
--------------------------------------------------------------
<?xml version="1.0" ?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <latitude_deg>47.3667</latitude_deg>
      <longitude_deg>8.5500</longitude_deg>
      <elevation>500.0</elevation>
      <heading_deg>0</heading_deg>
    </spherical_coordinates>
    <physics type='ode'>
      <ode>
        <solver>
          <type>quick</type>
          <iters>1000</iters>
          <sor>1.3</sor>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.01</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>100</real_time_update_rate>
      <gravity>0 0 -9.8</gravity>
    </physics>
  </world>
</sdf>
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------
task2_world.launch from BebopS
----------------------------------------------------
<?xml version="1.0"?>

<launch>
    <arg name="name" default="bebop"/>
    <arg name="plotName" default="bebopPlots"/>
    <arg name="x" default="0.0" />
    <arg name="y" default="0.0" />
    <arg name="z" default="0.0" />
    <arg name="use_sim_time" default="true"/>
    <arg name="world_name" default="basic1"/>
    <arg name="wind_force" default="0.25"/>
    <arg name="wind_start" default="5.0"/>
    <arg name="wind_duration" default="40.0"/>
    <arg name="wind_direction_x" default="1.0"/>
    <arg name="wind_direction_y" default="0.0"/>
    <arg name="wind_direction_z" default="0.0"/>
    <arg name="csvFilesStoring" default="false"/>
    <arg name="csvFilesStoringTime" default="60.0"/> 
    <arg name="user_account" default="giuseppe"/> 
    <arg name="waypoint_filter" default="true"/>
    <arg name="EKFActive" default="false"/>
    <arg name="enable_odometry_sensor_with_noise" default="false"/>
    <arg name="disable_odometry_sensor_with_noise" default="true"/>
    <arg name="enable_ground_truth_sensor" default="false"/>
    <arg name="enable_wind_plugin" default="false"/>
    <arg name="enable_laser1D" default="false"/>
[... 79 lines cut ...]
    
    <node name="uTerr_components_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/uTerrComponents/pose/pose/position" output="screen" />

    
    <node name="zVelocity_components_plot_1" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/zVelocityComponents/pose/pose/position" output="screen" />
    <node name="zVelocity_components_plot_2" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/zVelocityComponents/twist/twist/linear" output="screen" />

    
    <node name="position_plot_ekf" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/filteredOutput/pose/pose/position" output="screen" />
    <node name="velocity_plot_ekf" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/filteredOutput/twist/twist/linear" output="screen" />

    
    <node name="state_errors_position_plot_ekf" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/stateErrors/pose/pose/position" output="screen" />
    <node name="state_errors_velocity_plot_ekf" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/stateErrors/twist/twist/linear" output="screen" />
    
    
    <node name="position_errors_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/positionAndVelocityErrors/pose/pose/position" output="screen" />
    <node name="velocity_errors_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/positionAndVelocityErrors/twist/twist/linear" output="screen" />

    
    <node name="angular_errors_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/angularAndAngularVelocityErrors/pose/pose/position" output="screen" />
    <node name="angular_velocity_errors_plot" pkg="rqt_plot" type="rqt_plot" args="--clear-config /$(arg name)/angularAndAngularVelocityErrors/twist/twist/linear" output="screen" />
   </group>

</launch>

golden-memory, 2020-07-30:
basic1.world
-----------------------------------------------------------------------------------
<?xml version="1.0" ?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <latitude_deg>47.3667</latitude_deg>
      <longitude_deg>8.5500</longitude_deg>
      <elevation>500.0</elevation>
      <heading_deg>0</heading_deg>
    </spherical_coordinates>
    <physics type='ode'>
      <ode>
        <solver>
          <type>quick</type>
          <iters>1000</iters>
          <sor>1.3</sor>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.01</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>100</real_time_update_rate>
      <gravity>0 0 -9.8</gravity>
    </physics>
  </world>
</sdf>

golden-memory, 2020-07-30:
basic1.world
`<?xml version="1.0" ?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <latitude_deg>47.3667</latitude_deg>
      <longitude_deg>8.5500</longitude_deg>
      <elevation>500.0</elevation>
      <heading_deg>0</heading_deg>
    </spherical_coordinates>
    <physics type='ode'>
      <ode>
        <solver>
          <type>quick</type>
          <iters>1000</iters>
          <sor>1.3</sor>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.01</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>100</real_time_update_rate>
      <gravity>0 0 -9.8</gravity>
    </physics>
  </world>
</sdf>`

golden-memory, 2020-07-30:
basic1.world
============================================
<?xml version="1.0" ?>
<sdf version="1.4">
  <world name="default">
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <include>
      <uri>model://sun</uri>
    </include>
    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <latitude_deg>47.3667</latitude_deg>
      <longitude_deg>8.5500</longitude_deg>
      <elevation>500.0</elevation>
      <heading_deg>0</heading_deg>
    </spherical_coordinates>
    <physics type='ode'>
      <ode>
        <solver>
          <type>quick</type>
          <iters>1000</iters>
          <sor>1.3</sor>
        </solver>
        <constraints>
          <cfm>0</cfm>
          <erp>0.2</erp>
          <contact_max_correcting_vel>100</contact_max_correcting_vel>
          <contact_surface_layer>0.001</contact_surface_layer>
        </constraints>
      </ode>
      <max_step_size>0.01</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>100</real_time_update_rate>
      <gravity>0 0 -9.8</gravity>
    </physics>
  </world>
</sdf>

golden-memory, 2020-07-30:
basic1.world
-----------------------------------------------------------------------------------------------------------------
[basic1.txt](https://github.com/gsilano/BebopS/files/5000164/basic1.txt)

golden-memory, 2020-07-30:
task2_world.launch
--------------------------------------------------------
[task2_world.launch.txt](https://github.com/gsilano/BebopS/files/5000176/task2_world.launch.txt)

golden-memory, 2020-07-30:
task2_world.launch
[task2_world.launch.txt](https://github.com/gsilano/BebopS/files/5000197/task2_world.launch.txt)

golden-memory, 2020-07-30:
<arg name="world_name" default="basic2"/>, basic1! not basic2

golden-memory, 2020-07-30:
in task2_world.launch, where <arg name="world_name" default="basic2"/> , basic1 but basic2

golden-memory, 2020-07-31:
> Copy and paste the content of the launch and world file

I thank you very much, now the drone can fly well!
[image]

closed 2020-08-02 by gsilano