gsilano/BebopS #27: Multi drone flight

golden-memory, 2020-08-04:
I want to make multiple drones fly, but i don't know how to do it!
I run the command in a terminal as following
`roslaunch bebop_simulator fly1.launch`
[image]
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
fly1.launch
'''xml
<?xml version="1.0"?>

<launch>
    <arg name="name" default="bebop"/>
    <arg name="plotName" default="bebopPlots"/>
    <arg name="use_sim_time" default="true"/>
    <arg name="world_name" default="test_city"/>
    <arg name="wind_force" default="0.25"/>
    <arg name="wind_start" default="5.0"/>
    <arg name="wind_duration" default="40.0"/>
    <arg name="wind_direction_x" default="1.0"/>
    <arg name="wind_direction_y" default="0.0"/>
    <arg name="wind_direction_z" default="0.0"/>
    <arg name="csvFilesStoring" default="false"/>
    <arg name="csvFilesStoringTime" default="60.0"/>
    <arg name="user_account" default="giuseppe"/> 
    <arg name="waypoint_filter" default="true"/>
    <arg name="EKFActive" default="false"/>
    <arg name="enable_odometry_sensor_with_noise" default="false"/>
    <arg name="disable_odometry_sensor_with_noise" default="true"/>
    <arg name="enable_ground_truth_sensor" default="false"/>
    <arg name="enable_wind_plugin" default="false"/>
    <arg name="enable_laser1D" default="false"/>
    <arg name="enable_plots" default="false"/>
    <arg name="enable_imu" default="true"/>

    
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find bebop_simulator)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find bebop_simulator)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find bebop_simulator)/worlds/$(arg world_name).world" />
  </include>

    <group ns="firefly">
        <include file="$(find bebop_simulator)/launch/spawn_bebop.launch">
           <arg name="name" value="$(arg name)" />
           <arg name="x" value="0.0"/>
           <arg name="y" value="0.0"/>
           <arg name="z" value="0.5"/>
           <arg name="wind_force" value="$(arg wind_force)"/>
           <arg name="wind_start" value="$(arg wind_start)"/>
           <arg name="wind_duration" value="$(arg wind_duration)"/>
           <arg name="wind_direction_x" value="$(arg wind_direction_x)"/>
           <arg name="wind_direction_y" value="$(arg wind_direction_y)"/>
           <arg name="wind_direction_z" value="$(arg wind_direction_z)"/>
           <arg name="enable_odometry_sensor_with_noise" value="$(arg enable_odometry_sensor_with_noise)"/>
           <param name="disable_odometry_sensor_with_noise" value="false" unless="$(arg enable_odometry_sensor_with_noise)" />
           <arg name="disable_odometry_sensor_with_noise" value="$(arg disable_odometry_sensor_with_noise)"/>
           <arg name="enable_ground_truth_sensor" value="$(arg enable_ground_truth_sensor)"/>
           <arg name="enable_wind_plugin" value="$(arg enable_wind_plugin)"/>
           <arg name="enable_imu" value="$(arg enable_imu)"/>
           <arg name="enable_laser1D" value="$(arg enable_laser1D)"/>
        </include>

        <node name="position_controller_node" pkg="bebop_simulator" type="position_controller_node" output="screen"> 
          <rosparam command="load" file="$(find bebop_simulator)/resource/controller_bebop.yaml" />
          <rosparam command="load" file="$(find bebop_simulator)/resource/bebop.yaml" />
          <rosparam command="load" file="$(find bebop_simulator)/resource/EKF_matrix.yaml" />
          <rosparam command="load" file="$(find bebop_simulator)/resource/waypoint_filter.yaml" />
      	  <param name="use_sim_time" value="$(arg use_sim_time)" />
          <param name="csvFilesStoring" value="$(arg csvFilesStoring)"/>
          <param name="csvFilesStoringTime" value="$(arg csvFilesStoringTime)"/>
          <param name="user_account" value="$(arg user_account)"/>
[... 2160 lines cut ...]
          </geometry>
          <material>
            <script>
              <uri>model://gas_station/materials/scripts</uri>
              <uri>model://gas_station/materials/textures</uri>
              <name>GasStation/Diffuse</name>
            </script>
            <shader type='normal_map_object_space'>
              <normal_map>GasStation_Normal.png</normal_map>
            </shader>
          </material>
        </visual>
        <velocity_decay>
          <linear>0</linear>
          <angular>0</angular>
        </velocity_decay>
        <self_collide>0</self_collide>
        <kinematic>0</kinematic>
        <gravity>1</gravity>
      </link>
      <pose>25 -13 0 0 0 -1.5708</pose>
    </model>
  </world>
</sdf>
'''

gsilano, 2020-08-04:
Please, this issue: https://github.com/gsilano/CrazyS/issues/44 Although it refers to CrazyS (another work of mine), the procedure should be exactly the same.

golden-memory, 2020-08-05:
> Please, this issue: [gsilano/CrazyS#44](https://github.com/gsilano/CrazyS/issues/44) Although it refers to CrazyS (another work of mine), the procedure should be exactly the same.

Thanks for your response. Now, I know that the problem may be related to the namespace. But I don't know how to modify it, can you give me some guidance?
Firstly, the original launch file and rqt_graph are here:
```xml
<?xml version="1.0"?>

<launch>
    <arg name="name" default="bebop"/>
    <arg name="plotName" default="bebopPlots"/>
    <arg name="world_name" default="test_city"/>
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find bebop_simulator)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find bebop_simulator)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find bebop_simulator)/worlds/$(arg world_name).world" />
  </include>

    <include file="$(find bebop_simulator)/launch/spawn_bebop.launch">
       <arg name="name" value="$(arg name)" />
       <arg name="x" value="0.0"/>
       <arg name="y" value="0.0"/>
       <arg name="z" value="0.3"/>
       <arg name="wind_force" value="0.0"/>
       <arg name="wind_start" value="0.0"/>
       <arg name="wind_duration" value="40.0"/>
       <arg name="wind_direction_x" value="0.0"/>
       <arg name="wind_direction_y" value="0.0"/>
       <arg name="wind_direction_z" value="0.0"/>
       <arg name="enable_odometry_sensor_with_noise" value="false"/>
       <arg name="disable_odometry_sensor_with_noise" value="true"/>
       <arg name="enable_ground_truth_sensor" value="false"/>
       <arg name="enable_wind_plugin" value="false"/>
       <arg name="enable_imu" value="true"/>
       <arg name="enable_laser1D" value="false"/>
    </include>

    <node name="position_controller_node" pkg="bebop_simulator" type="position_controller_node" output="screen"> 
      <rosparam command="load" file="$(find bebop_simulator)/resource/controller_bebop.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/bebop.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/EKF_matrix.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/waypoint_filter.yaml" />
      <param name="use_sim_time" value="true" />
      <param name="csvFilesStoring" value="false"/>
      <param name="csvFilesStoringTime" value="60.0"/>
      <param name="user_account" value="giuseppe"/>
      <param name="waypoint_filter" value="true"/>
      <param name="EKFActive" value="false"/>
      <remap from="/command/motor_speed" to="/gazebo/command/motor_speed" />
      <remap from="/odometry" to="/bebop/odometry" />
      <remap from="/odometry_gt" to="/bebop/odometry_gt" />
      <remap from="/filteredOutput" to="/bebop/filteredOutput" />
      <remap from="/referenceAngles" to="/bebop/referenceAngles" />
      <remap from="/stateErrors" to="/bebop/stateErrors" />
      <remap from="/smoothedTrajectory" to="/bebop/smoothedTrajectory" />
      <remap from="/command/trajectory" to="/bebop/command/trajectory" />
      <remap from="/uTerrComponents" to="/bebop/uTerrComponents" />
      <remap from="/zVelocityComponents" to="/bebop/zVelocityComponents" />
      <remap from="/positionAndVelocityErrors" to="/bebop/positionAndVelocityErrors" />
      <remap from="/angularAndAngularVelocityErrors" to="/bebop/angularAndAngularVelocityErrors" />
    </node>

    <!-- Launch the trajectory generator -->
   <group ns="$(arg name)">
       <node name="waypoint_example" pkg="bebop_simulator" type="waypoint_example" output="screen" args="$(find bebop_simulator)/resource/waypoints.txt"/>   
    </group>

</launch>
```
[image]
-------------------------------------------------------------------------------------------------------------------------------------------------------
[... 190 lines cut ...]
          <param name="waypoint_filter" value="true"/>
          <param name="EKFActive" value="false"/>
          <remap from="/command/motor_speed" to="/gazebo/command/motor_speed"/>
          <remap from="/odometry" to="/bebop/odometry" />
          <remap from="/odometry_gt" to="/bebop/odometry_gt" />
          <remap from="/filteredOutput" to="/bebop/filteredOutput" />
          <remap from="/referenceAngles" to="/bebop/referenceAngles" />
          <remap from="/stateErrors" to="/bebop/stateErrors" />
          <remap from="/smoothedTrajectory" to="/bebop/smoothedTrajectory" />
          <remap from="/command/trajectory" to="/bebop/command/trajectory" />
          <remap from="/uTerrComponents" to="/bebop/uTerrComponents" />
          <remap from="/zVelocityComponents" to="/bebop/zVelocityComponents"/>
          <remap from="/positionAndVelocityErrors" to="/bebop/positionAndVelocityErrors" />
          <remap from="/angularAndAngularVelocityErrors" to="/bebop/angularAndAngularVelocityErrors" />
        </node>
     </group>

    <!-- change the namespace -->
   <group ns="$(arg name)1">
       <node name="waypoint_example" pkg="bebop_simulator" type="waypoint_example" output="screen" args="$(find bebop_simulator)/resource/waypoints.txt"/>   
     </group>

</launch>
```
[image]

gsilano, 2020-08-05:
I suggest you starting [from this file](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/crazyflie2_swarm_hovering_example.launch), and then taking a look at all the related file (e.g., the [spawn_mav_crazyfle](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/spawn_mav_crazyflie.launch)). I hope this helps.

golden-memory, 2020-08-10:
> I suggest you start [from this file](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/crazyflie2_swarm_hovering_example.launch), and then take a look at all the related file (e.g., the [spawn_mav_crazyfle](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/spawn_mav_crazyflie.launch)). I hope this helps.

I'm so sorry to disturb you again. But now, I have no idea about the problem. My recent study status is as follows:
----------------------------------------------------------------------------------------------------------------------------------------------
Firstly, The biggest problem I have encountered is that **gazebo** node and **position_controller_node** node cannot communicate.
The correct graph should be:
[image]
But my graph is:
[image]
------------------------------------------------------------------------------------------------------------------------------------------
Secondly, I tried to make some modifications, but it still doesn't work.
The file I modified is [spawn_bebop.launch](https://github.com/gsilano/BebopS/blob/master/launch/spawn_bebop.launch). On line 34, I change **namespace:=$(arg name)** to amespace:='$(arg name)_1' where the first drone is located. Unfortunately it reported an error:
[image]
-----------------------------------------------------------------------------------------------------------------------------------------------------
At last, Related launch files are as follows:
fly6.launch
```xlm
<?xml version="1.0"?>

<launch>
    <arg name="name" default="bebop"/>
    <arg name="plotName" default="bebopPlots"/>
    <arg name="world_name" default="test_city"/>
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find bebop_simulator)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find bebop_simulator)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find bebop_simulator)/worlds/$(arg world_name).world" />
  </include>

    <!-- add the namespace -->
    <group ns="$(arg name)_1">
        <include file="$(find bebop_simulator)/launch/spawn_bebop.launch">
           <arg name="name" value="$(arg name)"/>
           <arg name="x" value="0.0"/>
           <arg name="y" value="0.0"/>
           <arg name="z" value="0.3"/>
           <arg name="wind_force" value="0.0"/>
           <arg name="wind_start" value="0.0"/>
           <arg name="wind_duration" value="40.0"/>
           <arg name="wind_direction_x" value="0.0"/>
           <arg name="wind_direction_y" value="0.0"/>
           <arg name="wind_direction_z" value="0.0"/>
           <arg name="enable_odometry_sensor_with_noise" value="false"/>
           <arg name="disable_odometry_sensor_with_noise" value="true"/>
           <arg name="enable_ground_truth_sensor" value="false"/>
           <arg name="enable_wind_plugin" value="false"/>
           <arg name="enable_imu" value="true"/>
           <arg name="enable_laser1D" value="false"/>
        </include>

        <node name="position_controller_node" pkg="bebop_simulator" type="position_controller_node" output="screen"> 
          <rosparam command="load" file="$(find bebop_simulator)/resource/controller_bebop.yaml" />
          <rosparam command="load" file="$(find bebop_simulator)/resource/bebop.yaml" />
          <rosparam command="load" file="$(find bebop_simulator)/resource/EKF_matrix.yaml" />
          <rosparam command="load" file="$(find bebop_simulator)/resource/waypoint_filter.yaml" />
          <param name="use_sim_time" value="true" />
          <param name="csvFilesStoring" value="false"/>
          <param name="csvFilesStoringTime" value="60.0"/>
          <param name="user_account" value="giuseppe"/>
          <param name="waypoint_filter" value="true"/>
          <param name="EKFActive" value="false"/>
          <remap from="/command/motor_speed" to="/gazebo/command/motor_speed"/>
          <remap from="/odometry" to="/bebop/odometry" />
          <remap from="/odometry_gt" to="/bebop/odometry_gt" />
          <remap from="/filteredOutput" to="/bebop/filteredOutput" />
          <remap from="/referenceAngles" to="/bebop/referenceAngles" />
          <remap from="/stateErrors" to="/bebop/stateErrors" />
          <remap from="/smoothedTrajectory" to="/bebop/smoothedTrajectory" />
          <remap from="/command/trajectory" to="/bebop/command/trajectory" />
          <remap from="/uTerrComponents" to="/bebop/uTerrComponents" />
[... 48 lines cut ...]
    wind_duration:=$(arg wind_duration)
    wind_direction_x:=$(arg wind_direction_x)
    wind_direction_y:=$(arg wind_direction_y)
    wind_direction_z:=$(arg wind_direction_z)
    "
  />
  <param name="tf_prefix" type="string" value="$(arg tf_prefix)" />

  <!-- push robot_description to factory and spawn robot in gazebo -->
  <node name="spawn_robot" pkg="gazebo_ros" type="spawn_model"
   args="-param robot_description
         -urdf
         -x $(arg x)
         -y $(arg y)
         -z $(arg z)
         -model $(arg name)"
   respawn="false" output="screen" >
  </node>

</launch>
```
The command entered is
`roslaunch bebop_simulator fly6.launch`
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
My question is a bit stupid, sorry to disturb you.

gsilano, 2020-08-10:
Okay, I get it. Let's do it this way. To save time, I'll create this file for you and then upload it to the repository. I should give back to you in a few days.

golden-memory, 2020-08-11:
> Okay, I get it. Let's do it this way. To save time, I'll create this file for you and then upload it to the repository. I should give back to you in a few days.

Thank you so much！

golden-memory, 2020-08-24:
@gsilano Thank you for your reply!  
Firstly, we need to fix some minor errors. In `task1_world.launch` ([https://github.com/gsilano/BebopS/blob/master/launch/task1_world.launch](url)), I deleted the 29 line which is the same as the 20 line. In the file you gave above, we need to change name_1 to name_2 in 4 line. 
Secondly, I have run the `task1_world.launch`, Although there is an error, the drone is indeed flying. Then I ran the file you gave above, The following error occurred
`roslaunch file contains multiple nodes named [/spawn_robot].
Please check all <node> 'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file
`
Besides, can you tell me where to modify the gazebo node？In some modifications, I found that it did not publish a message！
[image]
[image]

golden-memory, 2020-08-24:
Or where is the `/bebop/odometry`, Is it not a variable？

gsilano, 2020-08-25:
Mhmm. As I can understand, add a second Bebop is not so easy as I supposed to be. This requires a lot of effort and a redesign of the simulation framework. The easiest way could be to start from the [multi-robot example provided in the CrazyS repository](https://github.com/gsilano/CrazyS/blob/master/rotors_gazebo/launch/crazyflie2_swarm_hovering_example.launch), replacing the .xacro files used for the Crazyflie with those of the Bebop. This also means that you should manage the moving of the control algorithm.

golden-memory, 2020-08-26:
@gsilano
I thank you from the bottom of my heart！Now drones can fly well, the problem is basically solved. As a beginner, I am very satisfied! And it all comes from your help to me, thanks ever so!

closed 2020-08-26 by gsilano