gsilano/BebopS #30: The rotors don't spin

sunnyshi0310, 2020-09-20:
Hi, after installing the BebopS, I run the commend
`$ roslaunch bebop_simulator bebop_without_controller.launch`
The quadrotor shows successfully, while when I run
`$ rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'`
The terminal shows
`publishing and latching message`
while the rotors of the aerial-robot don't spin.  

Then I tried the 
`$ roslaunch bebop_simulator task1_world.launch`
it shows the error
`Invalid <tag> tag: arg 'enable_imu' has already been declared.`
`Arg xml is <arg default="true" name="enable_imu"/>`

It will be very kind of you if anyone can help me with that. Thanks!!!

gsilano, 2020-09-20:
I suggest starting from #26, #25, #23, #19, and #4. If the problem still persists, please let me know.

sunnyshi0310, 2020-09-20:
Hi, thanks for your reply. I have checked those issues but unfortunately they didn't help. I attached the screenshots when I tried to run
`roslaunch bebop_simulator bebop_without_controller.launch`
[image]
[image]
[image]
Everything seems work well but the rotors don't spin.
btw, I work with Ubuntu 16.04 and Gazebo 7. Thanks in advance.

gsilano, 2020-09-21:
As you can see, the `gazebo/command/motor_speed` topic does not have a connection with Gazebo. For this reason, the motors do not spin. Try to publish the same command on the `bebop/command/motors` or `/command/motor_speed` topic, and let me know.
Also, could you send me the list of the available topics? I think there may be a typo in the README or in the launch file.

sunnyshi0310, 2020-09-21:
Hi, thanks for your reply. I got three issues.
- When I try the `roslaunch bebop_simulator bebop_without_controller.launch` I publish the command on the `bebop/command/motors` topic  and the quadrotor starts to run, but it runs randomly and dropped soon. So I decrease the angular velocity from 1000 to 100 and it stays at the origin with spinning rotors. So I think the problem is solved.  :) That's its rqt_graph and rostopic list.I also tried it on the `/command/motor_speed` topic and the rotors don't spin. 
[image]
[image]

- I try the `roslaunch bebop_simulator task1_world.launch`. It shows the error
[image]
- I run the  `roslaunch bebop_simulator task1_world.launch`. The rotors don't spin and the robot stays in the origin. Its rqt_graph and rostopic list are shown as follows
[image]
[image]
[image]

Thanks again for your work.!!

gsilano, 2020-09-21:
So, great. This means it works. I mean, we found the motor spin issue. Obviously, automatic control feedback is needed to control system behavior.

I just removed the typo. Please, try it again. Try to replace this line `<remap from="/command/motor_speed" to="/gazebo/command/motor_speed" />` with this `` (simply comment it), and let me know.

sunnyshi0310, 2020-09-21:
Hi, I comment the line `<arg name="enable_imu" default="true"/>` and run the `roslaunch bebop_simulator task1_world.launch
`, while errors occur as follows:
[image]

gsilano, 2020-09-22:
Replace the content of the file with this

```xml
<?xml version="1.0"?>
<launch>
    <arg name="name" default="bebop"/>
    <arg name="x" default="0.0" />
    <arg name="y" default="0.0" />
    <arg name="z" default="0.02" />
    <arg name="use_sim_time" default="true"/>
    <arg name="world_name" default="basic"/>
    <arg name="wind_force" default="0.5"/>
    <arg name="wind_start" default="15.0"/>
    <arg name="wind_duration" default="10.0"/>
    <arg name="wind_direction_x" default="0.0"/>
    <arg name="wind_direction_y" default="0.0"/>
    <arg name="wind_direction_z" default="1.0"/>
    <arg name="csvFilesStoring" default="false"/>
    <arg name="csvFilesStoringTime" default="15.0"/> <!-- seconds -->
    <arg name="user_account" default="giuseppe"/>
    <arg name="waypoint_filter" default="true"/>
    <arg name="EKFActive" default="false"/>
    <arg name="enable_imu" default="true"/>
    <arg name="enable_odometry_sensor_with_noise" default="false"/>
    <!-- The "disable_odometry_sensor_with_noise" variable values is equal to true if "enable_odometry_sensor_with_noise"
    is false, and viceversa -->
    <arg name="disable_odometry_sensor_with_noise" value="true" unless="$(arg enable_odometry_sensor_with_noise)"/>
    <arg name="disable_odometry_sensor_with_noise" value="false" if="$(arg enable_odometry_sensor_with_noise)"/>
    <arg name="enable_ground_truth_sensor" default="false"/>
    <arg name="enable_wind_plugin" default="false"/>
    <arg name="enable_laser1D" default="false"/>
    
  <env name="GAZEBO_MODEL_PATH" value="${GAZEBO_MODEL_PATH}:$(find bebop_simulator)/models"/>
  <env name="GAZEBO_RESOURCE_PATH" value="${GAZEBO_RESOURCE_PATH}:$(find bebop_simulator)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find bebop_simulator)/worlds/$(arg world_name).world" />
  </include>

    <!-- Launch UAV -->
    <include file="$(find bebop_simulator)/launch/spawn_bebop.launch">
       <arg name="name" value="$(arg name)" />
       <arg name="x" value="$(arg x)"/>
       <arg name="y" value="$(arg y)"/>
       <arg name="z" value="$(arg z)"/>
       <arg name="wind_force" value="$(arg wind_force)"/>
       <arg name="wind_start" value="$(arg wind_start)"/>
       <arg name="wind_duration" value="$(arg wind_duration)"/>
       <arg name="wind_direction_x" value="$(arg wind_direction_x)"/>
       <arg name="wind_direction_y" value="$(arg wind_direction_y)"/>
       <arg name="wind_direction_z" value="$(arg wind_direction_z)"/>
      <!-- The disable_odometry_sensor_with_noise will only become true if enable_odometry_sensor_with_noise will is false.
       In this way, only one odometry sensor will be simulated: with or without noise. -->
       <arg name="enable_odometry_sensor_with_noise" value="$(arg enable_odometry_sensor_with_noise)"/>
       <arg name="disable_odometry_sensor_with_noise" value="$(arg disable_odometry_sensor_with_noise)" /> 
       <arg name="enable_ground_truth_sensor" value="$(arg enable_ground_truth_sensor)"/>
       <arg name="enable_wind_plugin" value="$(arg enable_wind_plugin)"/>
       <arg name="enable_laser1D" value="$(arg enable_laser1D)"/>
       <arg name="enable_imu" value="$(arg enable_imu)"/>
    </include>

    <!-- Launch the controller  -->
    <node name="position_controller_node" pkg="bebop_simulator" type="position_controller_node" output="screen"> 
      <rosparam command="load" file="$(find bebop_simulator)/resource/controller_bebop.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/bebop.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/EKF_matrix.yaml" />
      <rosparam command="load" file="$(find bebop_simulator)/resource/waypoint_filter.yaml" />
      <param name="use_sim_time" value="$(arg use_sim_time)" />
      <param name="csvFilesStoring" value="$(arg csvFilesStoring)"/>
      <param name="csvFilesStoringTime" value="$(arg csvFilesStoringTime)"/>
      <param name="user_account" value="$(arg user_account)"/>
      <param name="waypoint_filter" value="$(arg waypoint_filter)"/>
      <param name="EKFActive" value="$(arg EKFActive)"/>
      <remap from="/odometry" to="/bebop/odometry" />
      <remap from="/odometry_gt" to="/bebop/odometry_gt" />
      <remap from="/referenceAngles" to="/bebop/referenceAngles" />
      <remap from="/filteredOutput" to="/bebop/filteredOutput" />
      <remap from="/stateErrors" to="/bebop/stateErrors" />
      <remap from="/smoothedTrajectory" to="/bebop/smoothedTrajectory" />
      <remap from="/command/trajectory" to="/bebop/command/trajectory" />
    </node>

    <!-- Launch the trajectory generator -->
   <group ns="$(arg name)">
       <node name="hovering_example" pkg="bebop_simulator" type="hovering_example" output="screen"/>   
    </group>

</launch>
```

sunnyshi0310, 2020-09-22:
Hi, everything works well now in terms of the terminal. But the robot don't move when I run `roslaunch bebop_simulator task1_world.launch` or `roslaunch bebop_simulator task2_world.launch`. Should I do sth after run the launch file? Its topics and rqt_graph are listed here. I am really appreciated for your work and feel so sorry to bother you so much. :)
[image]
[image]
[image]

gsilano, 2020-09-24:
Run the `task1.launch` file, and then publish a message on `gazebo/command/motor_speed`. Let me know if this fix the issue. If so, I will update the launch file in the repository.

```
rostopic pub /gazebo/command/motor_speed mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}'
```

sunnyshi0310, 2020-09-25:
Hi, it performs same as I run `roslaunch bebop_simulator bebop_without_controller.launch`. 

- When I run `roslaunch bebop_simulator task1_world.launch` and then `rostopic pub /gabebop/command/motors mav_msgs/Actuators '{angular_velocities: [1000, 1000, 1000, 1000]}`, nothing happens. 

- Then I publish the commend on topic `/bebop/command/motors` , the robot starts to move (spins its rotors) but not achieve the objective of task 1 (Parrot Bebop takes off from the ground and keeps indefinitely the hovering position subjected to wind gusts (up to 0.5 N) for a minute. )

gsilano, 2020-09-25:
Please, pull my last change and try again.

sunnyshi0310, 2020-09-25:
Everything works now! Thanksssss!!!! btw, where is the code for landing? (or are there any command for landing?)

gsilano, 2020-09-25:
There is no command for landing.

closed 2020-09-25 by gsilano