gsilano/BebopS #31: SITL with QGC

a-tab-a, 2020-09-22:
Hello, can BebopS be connected with QGC and control from QGC? With simulation open in Gazebo, and utilizing  mavlink_bridge is it possible to connect to QGC? Or it needs additional interface or plugin?

gsilano, 2020-09-22:
I don't know if I understand what QGC stands for. However, if you want to control bebop with ROS, you have to use this package: [bebop_autonomy](https://github.com/AutonomyLab/bebop_autonomy).

a-tab-a, 2020-09-22:
I meant QGroundcontrol station or say any mission planner. How do I connect QGroundcontrol with BebopS? Thanks

gsilano, 2020-09-23:
I think this repository can help you in set up the environment: https://github.com/PX4/sitl_gazebo

closed 2021-01-23 by gsilano