gsilano/BebopS #32: Spawning Error

vvami, 2020-10-04:
Hello,

I keep getting an error whenever I try to run that launch command right after I install the package:
`roslaunch bebop_simulator bebop_without_controller.launch`

What I get: 
[rospack] Warning: error while crawling /home/ami/catkin_ws/src/BebopS: boost::filesystem::status: Too many levels of symbolic links: "/home/ami/catkin_ws/src/BebopS/BebopS"
[rospack] Warning: error while crawling /home/ami/catkin_ws/src/BebopS: boost::filesystem::status: Too many levels of symbolic links: "/home/ami/catkin_ws/src/BebopS/BebopS"
[rospack] Warning: error while crawling /home/ami/catkin_ws/src/BebopS: boost::filesystem::status: Too many levels of symbolic links: "/home/ami/catkin_ws/src/BebopS/BebopS"
[rospack] Warning: error while crawling /home/ami/catkin_ws/src/BebopS: boost::filesystem::status: Too many levels of symbolic links: "/home/ami/catkin_ws/src/BebopS/BebopS"
[ INFO] [1601810758.669281737]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1601810758.672245357]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1601810758.732420807]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1601810758.734325826]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
SpawnModel script started
[INFO] [1601810759.491823, 0.000000]: Loading model XML from ros parameter
[INFO] [1601810759.515599, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1601810759.831826863]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1601810759.839305712]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1601810759.855559, 0.000000]: Calling service /gazebo/spawn_urdf_model
[rospack] Warning: error while crawling /home/ami/catkin_ws/src/BebopS: boost::filesystem::status: Too many levels of symbolic links: "/home/ami/catkin_ws/src/BebopS/BebopS"
[rospack] Warning: error while crawling /home/ami/catkin_ws/src/BebopS: boost::filesystem::status: Too many levels of symbolic links: "/home/ami/catkin_ws/src/BebopS/BebopS"
[rospack] Error: package 'bebop_simulator' not found
[librospack]: error while executing command
[FATAL] [1601810760.056795514, 0.029000000]: Package[bebop_simulator] does not have a path
Service call failed: service [/gazebo/spawn_urdf_model] responded with an error:

I'm relatively new to ROS but I wanted to use this package to implement my control system on a drone for a school project. Any help would be greatly appreciated.

Thanks in advance.

gsilano, 2020-10-04:
Did you follow the instructions in the README file? There seems to be some problem with the `bashrc` file.

vvami, 2020-10-04:
First of all, thanks for your reply. To answer your question, yes I have followed all of the steps. What would you recommend to do in order to fix the problem with the bashrc file?

vvami, 2020-10-04:
Update: I have removed then re-installed Ubuntu 16.04 on a virtual machine. Then I re-installed ROS and Gazebo 7,  followed all the steps again, and was able to launch the simulation without any errors this time. I can work on designing/implementing my controller now. Thank you so much for providing such a beneficial package for simulation and testing purposes.

closed 2020-10-04 by vvami

mefool, 2021-04-19:
Hello, just wanted to inform that I also had a problem identical to this one, and was able to solve it by removing the link Bebop inside this repository (catkin_ws/src/BebopS/BebopS). 
Haven't tested with with all the launch files, but bebop_without_controller.launch and multiple_bebop_without_controller.launch, seem to work, while previously the UAVs did not spawn.

gsilano, 2021-04-20:
Hi @mefool! Thanks for sharing your solution.