gsilano/BebopS #36: Running {task1_world_with_bebop,task2_world_with_bebop}.launch files

Nelson89, 2020-12-09:
Hello,

I'm trying to run a task1_world_with_bebop.launch and task2_world_with_bebop.launch with command:
`$ roslaunch bebop_simulator task1_world_with_bebop.launch`. 

Unfortunately, problems arise when trying to run.
[image]

I don't know why this error is showing up. If I try use other launch files, it works fine. I have tried with: gazebo_bebop.launch, bebop_without_controller.launch, task1_world.launch, task2_world.launch, task3_world.launch, multiple_bebop_without_controller.launch, multiple_task2_world.launch.

I use ROS Melodic on Ubuntu 18.04 with Gazebo 9.0.0

gsilano, 2020-12-10:
Hi! The `with_bebop.launch` files consider the real Bebop in the loop. If you are only interested in simulating it, you can just run the standard version.

Nelson89, 2020-12-10:
Hi,
Thank you for your answer! So if I understand correctly, this version `with_bebop.launch` is about a real drone in the real world and the other version is about a gazebo simulation?

gsilano, 2020-12-10:
Yes, it is.

Nelson89, 2020-12-10:
Ok, now I understand it.
Thank you!

closed 2020-12-10 by Nelson89