gsilano/BebopS #39: Gzserver: symbol lookup error After Launching bebop_without_controller.launch

guiaugustoga, 2020-12-21:
Hi,

I'm using ubuntu 18.04 with ros melodic. I got the following output when i try to launch bebop_without_controller.launch :

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /tf_prefix: bebop
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    spawn_robot (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://localhost:11311

process[gazebo-1]: started with pid [4574]
process[gazebo_gui-2]: started with pid [4579]
process[spawn_robot-3]: started with pid [4584]
[ INFO] [1608589547.665981255]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1608589547.669707232]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1608589547.677100877]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1608589547.680358247]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1608589548.158183, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1608589548.166677, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1608589548.652894152]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1608589548.681106497]: Physics dynamic reconfigure ready.
[INFO] [1608589548.771771, 0.072000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1608589548.959109, 0.123000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1608589549.177460981, 0.123000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1608589549.180927510, 0.123000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to "bebop"
[spawn_robot-3] process has finished cleanly
log file: /home/gapc/.ros/log/8b5166b6-43da-11eb-a0d6-000516598047/spawn_robot-3*.log
gzserver: symbol lookup error: /home/gapc/catkin_bebop/devel/lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev
[gazebo-1] process has died [pid 4574, exit code 127, cmd /home/gapc/catkin_bebop/src/gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -e ode /home/gapc/catkin_bebop/src/BebopS/worlds/basic.world __name:=gazebo __log:=/home/gapc/.ros/log/8b5166b6-43da-11eb-a0d6-000516598047/gazebo-1.log].
log file: /home/gapc/.ros/log/8b5166b6-43da-11eb-a0d6-000516598047/gazebo-1*.log

I'm not sure what is causing this error. Thanks for your help !

gsilano, 2020-12-22:
This is similar to #34, #32, #30, #25, #21, #35, #37, and #27. Please, try to follow the described suggestions (see the content of the issue) and give me back if the problem still persists.

guiaugustoga, 2020-12-22:
I deleted my catkin and made all steps again with no sucess.

gsilano, 2020-12-22:
Are you able to run the standard setup? If so, you do not need to install other packages.

guiaugustoga, 2020-12-22:
What do you mean by standard setup ? I just followed the steps on readme.txt and couldn't launch bebop_without_controller.launch.

gsilano, 2020-12-22:
Okay, this means that something went wrong with it. Please, follow the instruction once more.

guiaugustoga, 2020-12-22:
I tried again and got the same error.

guiaugustoga, 2020-12-22:
[image]

Sometimes the gazebo opens even with the error message, but it is not running. The topics are also incomplete.

gsilano, 2020-12-22:
This seems to be an old error. Please, see the issues mentioned above.

guiaugustoga987, 2020-12-23:
I tried everything and it didn't work...

gsilano, 2020-12-23:
Let use a systematic approach to solve the problem. Describe me every single step precisely.

guiaugustoga987, 2020-12-23:
I solved my problem. I had to use catkin_make instead of catkin build. For this, I had to make some adjustments on package.xml and cmakelists.txt from rotors_simulator and then after building I was able to launch bebop_controller.launch with no errors.

Thanks for your attention !

closed 2020-12-23 by gsilano