gsilano/BebopS #40: bebop_autonomy with bebopS

guiaugustoga, 2020-12-29:
Hi,

Is there a way to use bebop_autonomy to control the drone on gazebo spawned by bebopS ?

Thanks !

gsilano, 2020-12-31:
To simulate the Parrot Bebop 2 in Gazebo accounting for the on-board firmware and aircraft behavior, I suggest you using [Spinx](https://developer.parrot.com/docs/sphinx/whatissphinx.html). As you can see from [this post](https://forum.developer.parrot.com/t/how-to-increase-the-output-frequency-of-the-data-logger-data/8758/3), we've already started working on it. If you take a look at the forum, you will find other useful info about the simulator, how to run it, and my code. See the branch `dev/Spinx`. The README.md file can help you understand more about it.

There is also the code you can use to control the real device. Indeed, the Sphinx allows you to use `bebop_autonomy` as an input interface.

I hope it helps.

closed 2021-06-26 by stale[bot]

jrthor07, 2023-11-28:
I don't understand, so your simulator doesn't support bebop autonomy? And the only thing it has in common with bebop is CAD?

gsilano, 2023-11-28:
The code released in the repository supports `bebop_autonomy`. For this, you should refer to the `dev/bebopinterface` and `dev/sphinx` branches