gsilano/BebopS #41: gzserver:symbol lookup error:

YunwooLee94, 2021-01-15:
Hello,

I am working on ubuntu18.04.
I tried to follow your install description (README). BebopS package was built without error, but, when I executed launch file, there was a problem. I copied a portion of error message.

gzserver: symbol lookup error: /opt/ros/melodic/share/gazebo_plugins/../../lib/librotors_gazebo_multirotor_base_plugin.so: undefined symbol: _ZN14gz_sensor_msgs9ActuatorsC1Ev

I found same issue, when I executed rotorS launch file.
Have you ever seen a message like this?

gsilano, 2021-01-17:
Hi @YunwooLee94! BebopS was developed upon RotorS, this is why you got the same error running the two ROS packages. This is the first time I see this error. It's been a long time since my last execution of the packages (now I'm working on other robotic simulators such as [MRS Simulator](https://github.com/ctu-mrs/mrs_simulation)), but I can suggest you (i) to check ROS installation by rerunning the installation process, (ii) if the `bashrc` file is set up correctly, (iii) run again all steps detailed in the installation procedure of BebopS being sure to run the `catkin clean` command first.

You could run the `catkin clean` command before going through the steps mentioned above. It may be a simple compiling error that reflects in the generation of links for the object file. I mean, try it's for free.

Please, do not hesitate to give it back to me if you have any other questions.

YunwooLee94, 2021-01-18:
Thank you for kind reply!!!
I finally solve the issue. (re-install ROS and follow your instruction)
BUT One different thing was that I used the original version of rotorS instead of branch "med18_gazebo9".
Thanks a lot.

closed 2021-01-18 by gsilano