gsilano/BebopS #43: Can't comple catkin_ws

EricDiazR, 2021-05-14:
Hello. 

I can't compile catkin_ws. It shows the following message:

-- Eigen found (include: /usr/include/eigen3)
-- +++ processing catkin package: 'rotors_gazebo'
-- ==> add_subdirectory(rotors_simulator/rotors_gazebo)
CMake Error at rotors_simulator/rotors_gazebo/CMakeLists.txt:27 (add_executable):
  add_executable cannot create target "hovering_example" because another
  target with the same name already exists.  The existing target is an
  executable created in source directory "/home/eric/catkin_ws/src/BebopS".
  See documentation for policy CMP0002 for more details.

CMake Error at rotors_simulator/rotors_gazebo/CMakeLists.txt:28 (target_link_libraries):
  Attempt to add link library "/opt/ros/melodic/lib/libroscpp.so" to target
  "hovering_example" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

-- Configuring incomplete, errors occurred!
See also "/home/eric/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/eric/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

I followed the instructions from the readme file. But i didn't clone the bebop_autonomy git cause i already had it (the anqixu fork that solves the compilation on ROS melodic)

Im running ubuntu 18.04 with ROS melodic

Thank you in advance.

gsilano, 2021-05-16:
Hi! We need to understand if the `bebop_autonomy` dependencies are causing the issue. Please, try excluding the package from the build. Simply, removing it from the folder containing all the ROS packages. Hence, compile and a simple example file. You should be able to install and use BebopS even without `bebop_autonomy`.

EricDiazR, 2021-05-18:
While using rosdep i get the following error message:
~/catkin_ws$ rosdep install --from-paths src -i

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bebop_simulator: Cannot locate rosdep definition for [bebop_driver]

And when trying to compile the catkin_ws i get the following (using catkin build)
Errors     << rotors_gazebo_plugins:make /home/eric/catkin_ws/logs/rotors_gazebo_plugins/build.make.000.log
MagneticField.proto: warning: Import quaternion.proto but not used.
/home/eric/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp: In member function ‘virtual void gazebo::GazeboOdometryPlugin::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
/home/eric/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp:94:48: error: ‘CV_LOAD_IMAGE_GRAYSCALE’ was not declared in this scope
     covariance_image_ = cv::imread(image_name, CV_LOAD_IMAGE_GRAYSCALE);
                                                ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rotors_gazebo_odometry_plugin.dir/src/gazebo_odometry_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_odometry_plugin.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
make: *** [all] Error 2
cd /home/eric/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]     
Failed    <<< rotors_gazebo_plugins                 [ 3 minutes and 20.7 seconds ]
Abandoned <<< bebop_simulator                       [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]   
[build] Summary: 9 of 12 packages succeeded.                                   
[build]   Ignored:   2 packages were skipped or are blacklisted.               
[build]   Warnings:  2 packages succeeded with warnings.                       
[build]   Abandoned: 2 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 6 minutes and 19.5 seconds total.        

After clone the official bebop_autonomy package, rosdep gives no error.

But i still get the previous error

Errors     << rotors_gazebo_plugins:make /home/eric/catkin_ws/logs/rotors_gazebo_plugins/build.make.001.log
/home/eric/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp: In member function ‘virtual void gazebo::GazeboOdometryPlugin::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
/home/eric/catkin_ws/src/rotors_simulator/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp:94:48: error: ‘CV_LOAD_IMAGE_GRAYSCALE’ was not declared in this scope
     covariance_image_ = cv::imread(image_name, CV_LOAD_IMAGE_GRAYSCALE);
                                                ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rotors_gazebo_odometry_plugin.dir/src/gazebo_odometry_plugin.cpp.o] Error 1
make[1]: *** [CMakeFiles/rotors_gazebo_odometry_plugin.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
make: *** [all] Error 2
cd /home/eric/catkin_ws/build/rotors_gazebo_plugins; catkin build --get-env rotors_gazebo_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << rotors_gazebo_plugins:make            [ Exited with code 2 ]     
Failed    <<< rotors_gazebo_plugins                 [ 20.2 seconds ]           
Abandoned <<< bebop_tools                           [ Unrelated job failed ]   
Abandoned <<< bebop_simulator                       [ Unrelated job failed ]   
Abandoned <<< rotors_gazebo                         [ Unrelated job failed ]

EricDiazR, 2021-05-18:
And if i use catkin_make instead of catkin build, i get this error (the same as the first comment of this issue):

-- Eigen found (include: /usr/include/eigen3)
CMake Error at BebopS/CMakeLists.txt:73 (add_executable):
  add_executable cannot create target "hovering_example" because another
  target with the same name already exists.  The existing target is an
  executable created in source directory
  "/home/eric/catkin_ws/src/rotors_simulator/rotors_gazebo".  See
  documentation for policy CMP0002 for more details.

CMake Error at BebopS/CMakeLists.txt:74 (target_link_libraries):
  Attempt to add link library "/opt/ros/melodic/lib/libroscpp.so" to target
  "hovering_example" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

-- Configuring incomplete, errors occurred!
See also "/home/eric/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/eric/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

gsilano, 2021-05-20:
Ok, got it. Remove `bebop_autonomy` from the `src` folder . Then, run again the `sudo rosdep` command, and finally `catkin build`. It should work.

SarahIchikawa, 2021-05-28:
Hello, I had the exact same error as @EricDiazR .
In my case, simply the version of OpenCV was different.
I suppose the [rotors_simulator](https://github.com/gsilano/rotors_simulator) code was written with OpenCV version lower than 3.0. (This might be useful: ['CV_LOAD_IMAGE_GRAYSCALE' is not defined{PY}](https://stackoverflow.com/questions/27424285/cv-load-image-grayscale-is-not-definedpy))
In my environment, I use ROS melodic + Ubuntu 18.04 + Gazebo 9 with **OpenCV v3.2**.

I had to;
-  add  `#include <opencv2/imgcodecs.hpp>`  
-  change from https://github.com/gsilano/rotors_simulator/blob/5c36c7f2affe42dbeabbf27601aeb0fc6fbc9093/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp#L94 to `    covariance_image_ = cv::imread(image_name, cv::ImreadModes::IMREAD_GRAYSCALE);` 

Modifying these solved the issue and I can compile packages now.

 @gsilano , original package with the latest version is doing the same thing. Please consider git pull from the original repository.
https://github.com/ethz-asl/rotors_simulator/blob/c1d4ef987b5b903fee0262524428b1b98c00a660/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp

SarahMars, 2021-05-28:
Hello, I had the exact same error as @EricDiazR .
In my case, simply the version of OpenCV was different.
I suppose the [rotors_simulator](https://github.com/gsilano/rotors_simulator) code was written with OpenCV version lower than 3.0. (This might be useful: ['CV_LOAD_IMAGE_GRAYSCALE' is not defined{PY}](https://stackoverflow.com/questions/27424285/cv-load-image-grayscale-is-not-definedpy))
In my environment, I use ROS melodic + Ubuntu 18.04 + Gazebo 9 with **OpenCV v3.2**.

I had to;
-  add  `#include <opencv2/imgcodecs.hpp>`  
-  change from https://github.com/gsilano/rotors_simulator/blob/5c36c7f2affe42dbeabbf27601aeb0fc6fbc9093/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp#L94 to `    covariance_image_ = cv::imread(image_name, cv::ImreadModes::IMREAD_GRAYSCALE);` 

Modifying these solved the issue and I can compile packages now.

 @gsilano , original package with the latest version is doing the same thing. Please consider git pull from the original repository.
https://github.com/ethz-asl/rotors_simulator/blob/c1d4ef987b5b903fee0262524428b1b98c00a660/rotors_gazebo_plugins/src/gazebo_odometry_plugin.cpp

<blockquote>[image]<div>[image] GitHub</div><div><strong><a href="https://github.com/gsilano/rotors_simulator">gsilano/rotors_simulator</a></strong></div><div>RotorS is a UAV gazebo simulator. Contribute to gsilano/rotors_simulator development by creating an account on GitHub.</div></blockquote>
<blockquote>[image]<div>[image] GitHub</div><div><strong><a href="https://github.com/ethz-asl/rotors_simulator">ethz-asl/rotors_simulator</a></strong></div><div>RotorS is a UAV gazebo simulator. Contribute to ethz-asl/rotors_simulator development by creating an account on GitHub.</div></blockquote>

gsilano, 2021-05-29:
Hi @SarahMars! Thanks for the info. I just created a new branch in my RotorS fork. I will update the README.md file with the instructions.

gsilano, 2022-04-17:
Closed for inactivity

closed 2022-04-17 by gsilano