gsilano/BebopS #44: Invalid parameter and Error with Param xml

adarshkosta, 2021-08-02:
Hi,

I was able to install and build everything properly without any errors.

However, when I try to run any of the launch files from bebop_simulator (bebop_without_controller.launch, task1_world.launch etc.) I get the following error.

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global symbol: pi
when processing file: /home/akosta/bebop_ws/src/rotors_simulator/rotors_description/urdf/component_snippets.xacro
included from: /home/akosta/bebop_ws/src/BebopS/urdf/bebop.urdf.xacro
Invalid parameter "measurement_delay"
when instantiating macro: imu_plugin_macro (/home/akosta/bebop_ws/src/rotors_simulator/rotors_description/urdf/component_snippets.xacro)
in file: /home/akosta/bebop_ws/src/BebopS/urdf/bebop.urdf.xacro
RLException: while processing /home/akosta/bebop_ws/src/BebopS/launch/spawn_bebop.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/akosta/bebop_ws/src/BebopS/urdf/bebop.urdf.xacro', 'enable_ground_truth:=true', 'enable_odometry_sensor_with_noise:=false', 'disable_odometry_sensor_with_noise:=true', 'enable_ground_truth_sensor:=false', 'enable_wind_plugin:=false', 'enable_laser1D:=false', 'enable_imu:=true', 'namespace:=bebop', 'wind_force:=0.5', 'wind_start:=15.0', 'wind_duration:=10.0', 'wind_direction_x:=0.0', 'wind_direction_y:=0.0', 'wind_direction_z:=1.0']] returned with code [2]. 

Param xml is <param command="     $(find xacro)/xacro '$(arg model)'     enable_ground_truth:='$(arg enable_ground_truth)'     enable_odometry_sensor_with_noise:='$(arg enable_odometry_sensor_with_noise)'     disable_odometry_sensor_with_noise:='$(arg disable_odometry_sensor_with_noise)'     enable_ground_truth_sensor:='$(arg enable_ground_truth_sensor)'     enable_wind_plugin:='$(arg enable_wind_plugin)'     enable_laser1D:='$(arg enable_laser1D)'     enable_imu:='$(arg enable_imu)'     namespace:=$(arg name)     wind_force:=$(arg wind_force)     wind_start:=$(arg wind_start)     wind_duration:=$(arg wind_duration)     wind_direction_x:=$(arg wind_direction_x)     wind_direction_y:=$(arg wind_direction_y)     wind_direction_z:=$(arg wind_direction_z)     " name="robot_description"/>
The traceback for the exception was written to the log file

Log file:
[roslaunch-P3630-1396.log](https://github.com/gsilano/BebopS/files/6920040/roslaunch-P3630-1396.log)

I am running on Ubuntu 18.04.5 with ROS Melodic and Gazebo 9.
Has this been observed previously? 
Thanks in advance!

gsilano, 2021-08-03:
Hi! I think you have not installed the repository correctly. You are using an old or incorrect branch of RotorS. This is the missing [file](https://github.com/gsilano/rotors_simulator/blob/med18_gazebo9/rotors_description/urdf/component_snippets.xacro) in your RotorS repository. The root of the file, i.e., `/home/akosta/bebop_ws/src/rotors_simulator/rotors_description/urdf/component_snippets.xacro`, is reported in the error log.

These steps should solve the issue:

1. Check the origin of the repository using the `git remote -v` command. Run it in the `RotorS` folder. In case it is not, delete the folder and clone it again using the `git clone -b med18_gazebo9 https://github.com/gsilano/rotors_simulator.git`.
2. If your origin is `https://github.com/gsilano/rotors_simulator.git`, switch to the right branch using the `git checkout med18_gazebo9` command.
3. Please, also check the origin of the `mav_comm` repository. Repeat operations 1 and 2 for it. Of course, instead of `rotors_simulator`, you find `mav_comm` in the link.

Let me know if this fixes it.

adarshkosta, 2021-08-03:
Hi,
I had the remote for `rotors_simulator` set to `med18_gazebo9_fix`. Changed it to `med18_gazebo`.
Also, had an issue with `parrot_arsdk` and `bebop_autonomy`. Had to soft link the `parrot_arsdk` installation from `/opt/ros/melodic/lib/parrot_arsdk` to `<$CATKIN_WS>/devel/lib/parrot_arsdk` and add it to $LD_LIBRARY_PATH.

Able to launch the simulator now.
Thanks!

closed 2021-08-03 by gsilano