gsilano/BebopS #45: [ERROR]

adbidwai, 2021-08-08:
Hello, Thanks for open sourcing this work. My system has ROS Melodic and Gazebo9. I am trying to build the packages by following the instructions on the README but I am encountering the following error, I am not able to debug based on the verbose.
Is there any way this issue can be resolved? I am posting my complete error log for reference : 
```
 adbidwai@adbidwai-g3-3579  ~/bebop_ws  catkin build
-------------------------------------------------------
Profile:                     default
Extending:             [env] /home/adbidwai/nus_ws/devel:/opt/ros/melodic
Workspace:                   /home/adbidwai/bebop_ws
-------------------------------------------------------
Build Space:        [exists] /home/adbidwai/bebop_ws/build
Devel Space:        [exists] /home/adbidwai/bebop_ws/devel
Install Space:      [unused] /home/adbidwai/bebop_ws/install
Log Space:         [missing] /home/adbidwai/bebop_ws/logs
Source Space:       [exists] /home/adbidwai/bebop_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '17' packages in 0.0 seconds.           
[build] Updating package table.                       
Starting  >>> catkin_tools_prebuild                   
Finished  <<< catkin_tools_prebuild                [ 3.6 seconds ]
Starting  >>> bebop_description                       
Starting  >>> bebop_msgs                              
Starting  >>> mav_msgs                                
Starting  >>> rotors_comm                             
Starting  >>> rotors_description                      
Starting  >>> rotors_evaluation                       
Finished  <<< bebop_description                    [ 3.7 seconds ]
Finished  <<< rotors_description                   [ 3.8 seconds ]
Finished  <<< rotors_evaluation                    [ 4.9 seconds ]
______________________________________________________
Warnings   << mav_msgs:cmake /home/adbidwai/bebop_ws/logs/mav_msgs/build.cmake.000.log
CMake Warning at /opt/ros/melodic/share/cmake_modules/cmake/Modules/FindEigen.cmake:62 (message):
  The FindEigen.cmake Module in the cmake_modules package is deprecated.

  Please use the FindEigen3.cmake Module provided with Eigen.  Change
  instances of find_package(Eigen) to find_package(Eigen3).  Check the
  FindEigen3.cmake Module for the resulting CMake variable names.

Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

cd /home/adbidwai/bebop_ws/build/mav_msgs; catkin build --get-env mav_msgs | catkin env -si  /home/adbidwai/.local/bin/cmake /home/adbidwai/bebop_ws/src/mav_comm/mav_msgs --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/adbidwai/bebop_ws/devel/.private/mav_msgs -DCMAKE_INSTALL_PREFIX=/home/adbidwai/bebop_ws/install; cd -
......................................................
______________________________________________________
Warnings   << rotors_comm:make /home/adbidwai/bebop_ws/logs/rotors_comm/build.make.000.log
WARNING: Package name "xMonsterCPG" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "xMonsterCPG" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
cd /home/adbidwai/bebop_ws/build/rotors_comm; catkin build --get-env rotors_comm | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................
Finished  <<< rotors_comm                          [ 8.0 seconds ]
______________________________________________________
Warnings   << mav_msgs:make /home/adbidwai/bebop_ws/logs/mav_msgs/build.make.000.log
WARNING: Package name "xMonsterCPG" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
WARNING: Package name "xMonsterCPG" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
[... 164 lines cut ...]
/opt/ros/melodic/include/dynamic_reconfigure/ConfigDescription.h:40:5: error: expected ‘(’ before ‘,’ token
     , dflt(_alloc)  {
     ^
make[2]: *** [CMakeFiles/bebop.dir/src/bebop.cpp.o] Error 1
make[1]: *** [CMakeFiles/bebop.dir/all] Error 2
make: *** [all] Error 2
cd /home/adbidwai/bebop_ws/build/bebop_driver; catkin build --get-env bebop_driver | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
......................................................
Failed     << bebop_driver:make                    [ Exited with code 2 ]
Failed    <<< bebop_driver                         [ 16.2 seconds ]
Abandoned <<< bebop_tools                          [ Unrelated job failed ]
Abandoned <<< rotors_gazebo_plugins                [ Unrelated job failed ]
Abandoned <<< bebop_simulator                      [ Unrelated job failed ]
Abandoned <<< rotors_gazebo                        [ Unrelated job failed ]
Finished  <<< rotors_control                       [ 26.2 seconds ]
[build] Summary: 10 of 15 packages succeeded.         
[build]   Ignored:   3 packages were skipped or are blacklisted.
[build]   Warnings:  6 packages succeeded with warnings.
[build]   Abandoned: 4 packages were abandoned.       
[build]   Failed:    1 packages failed.               
[build] Runtime: 48.9 seconds total.                  
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

TIA

gsilano, 2021-08-09:
Hi! Please, see #43, #40, and #24.

gsilano, 2022-04-17:
Closed for inactivity

closed 2022-04-17 by gsilano