gsilano/BebopS #47: motor_speed or cmd_vel

Huss94, 2023-01-04:
Hello,
When using a real bebop with bebop_autonomy, i got the topic cmd_vel.
in this simulation there is only the motor_speed topic.
How can I have the cmd_vel topic in the simulation ? 

Thank you for your help

gsilano, 2023-01-05:
Hi! Thanks for being interested in my work. It's been a long time since I've worked on it, so honestly, I can't remember. At that time, if I'm not mistaken, we created a branch (`dev/bebopinterface`) where we worked on some glue code to connect the controller tested in Gazebo (it uses RotorS as framework) and the hardware (Bebop 2). 

During an [MIT exchange program](http://catalog.mit.edu/mit/undergraduate-education/academic-research-options/independent-activities-period/), we also started working with the [Parrot Sphinx simulator](https://developer.parrot.com/docs/sphinx/index.html), see the `dev/mitProject2019` branch. and `dev/sphinx`. I'm sorry, if I couldn't help more.

Huss94, 2023-01-05:
Hi ! 
I didn't know about Sphinx and it will probably help me in my work.
Thank you for your awnser, I will check it out.

Huss94, 2023-01-07:
Hi !
Sphinx does no longer support bebop and bebop 2  drones.
here are the currently supported drones https://firmware.parrot.com/Versions/index.html.

I need to synchronize two parrot bebop 2 so that they follow each other, avoid obstacles. I need to use deep reinforcement learning to do that so I need to use a realistic simualtion. 
Maybe you can advise me something ?
I know it's been a long time since you've worked on this project but if you have advices, I gladly take them

Thank you for your help, have a good day

gsilano, 2023-01-08:
Well, it's okay if you use Gazebo and RotorS to simulate vehicle dynamics, hence BebopS. However, if you can't rely on external tools, such as an old release of Sphinx, to simulate the onboard controller and you can't use `bebop_autonomy` to control the drone, the only way I see is to make and to deploy a copy of the onboard controller yourself.

In other words, if you're using `bebop_autobomy` to pilot the drone in field experiments, you just need to integrate it with Gazebo. If not, you have to either make everything from scratch or Google around.

Huss94, 2023-01-09:
Thank you for your advices, they are very welcome. 
I don't think I could use sphinx, so as you said I will train the model on BebopS. 

Have a good day.

closed 2023-01-09 by Huss94