gsilano/BebopS #48: ROS noetic and gazebo 11

longvucao, 2023-05-07:
hi, can you update the guide for ROS noetic and gazebo 11

gsilano, 2023-05-07:
Hi! Sure. Thank you!

longvucao, 2023-05-07:
Hi, can you tell me when will you update? I am in dire need of UAV simulation for my project!

gsilano, 2023-05-07:
I'm on it. It should be ready by the end of the day.

longvucao, 2023-05-07:
Thanks a lot!

gsilano, 2023-05-07:
I am awaiting for build automation scripts.

gsilano, 2023-05-07:
@longvucao ,  GitHub Action doesn't work. Right now, I don't have time to investigate what's wrong. Try following the instructions in the repository. Next, let me know if it works for you.

longvucao, 2023-05-08:
@gsilano I followed your instructions but when I run catkin build it gives me the following error:
Errors     << rotors_control:make /home/vcl/Drone_4/Dr_ws/logs/rotors_control/build.make.000.log
In file included from /home/vcl/Drone_4/Dr_ws/src/rotors_simulator/rotors_control/include/rotors_control/lee_position_controller.h:27,
                 from /home/vcl/Drone_4/Dr_ws/src/rotors_simulator/rotors_control/src/library/lee_position_controller.cpp:21:
/home/vcl/Drone_4/Dr_ws/src/rotors_simulator/rotors_control/include/rotors_control/common.h:30:10: fatal error: rotors_control/parameters.h: No such file or directory
   30 | #include "rotors_control/parameters.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/lee_position_controller.dir/build.make:76: CMakeFiles/lee_position_controller.dir/src/library/lee_position_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1402: CMakeFiles/lee_position_controller.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
cd /home/vcl/Drone_4/Dr_ws/build/rotors_control; catkin build --get-env rotors_control | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
You can see the details in this picture:
[]([image])
Do you have any ideas to solve this problem?

longvucao, 2023-05-08:
@gsilano I followed your instructions but when I run catkin build it gives me the following error:
Errors << rotors_control:make /home/vcl/Drone_4/Dr_ws/logs/rotors_control/build.make.000.log
In file included from /home/vcl/Drone_4/Dr_ws/src/rotors_simulator/rotors_control/include/rotors_control/lee_position_controller.h:27,
from /home/vcl/Drone_4/Dr_ws/src/rotors_simulator/rotors_control/src/library/lee_position_controller.cpp:21:
/home/vcl/Drone_4/Dr_ws/src/rotors_simulator/rotors_control/include/rotors_control/common.h:30:10: fatal error: rotors_control/parameters.h: No such file or directory
30 | #include "rotors_control/parameters.h"
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/lee_position_controller.dir/build.make:76: CMakeFiles/lee_position_controller.dir/src/library/lee_position_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1402: CMakeFiles/lee_position_controller.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
cd /home/vcl/Drone_4/Dr_ws/build/rotors_control; catkin build --get-env rotors_control | catkin env -si /usr/bin/make --jobserver-auth=3,4; cd -
You can see the details in this picture:
[image]
Do you have any ideas to solve this problem?

gsilano, 2023-05-08:
It should be okay (fixed with #[44bb104](https://github.com/gsilano/rotors_simulator/commit/44bb1043fb0b5140069fcf686209cd958486bbe7))

longvucao, 2023-05-08:
@gsilano I think you should check the rotors_control package again because when I rebuild it it says the parameters_ros.h file is missing. I'm afraid that there may be more files missing

gsilano, 2023-05-08:
Did you git pull?

longvucao, 2023-05-08:
Oh sorry, i'm missing that file, but i think you should fix the tutorial because maybe i followed git clone -b dev/ros-noetic https://github.com/gsilano/rotors_simulator.git's tutorial you should be missing the other 2 files.
However, I still get an error in the package rotors_gazebo_plugins, specifically as follows:
[image]
I am really depressed with the package rotors_gazebo_plugins because it always fails in this package even though I follow different instructions.
So I'm very grateful to you if you can help me install this package!!
Thank's!!

longvucao, 2023-05-09:
@gsilano Oh sorry, i'm missing that file, but i think you should fix the tutorial because maybe i followed git clone -b dev/ros-noetic https://github.com/gsilano/rotors_simulator.git's tutorial you should be missing the other 2 files. However, I still get an error in the package rotors_gazebo_plugins, specifically as follows: [image] I am really depressed with the package rotors_gazebo_plugins because it always fails in this package even though I follow different instructions. So I'm very grateful to you if you can help me install this package!! Thank's!!

gsilano, 2023-05-09:
Check it out now! I made some changes with https://github.com/gsilano/rotors_simulator/commit/3bfaf59bb8b7c08317fa67c5885d4a25e7521e8a

gsilano, 2023-05-09:
The top screen, please. I'll try help if I can.

longvucao, 2023-05-09:
@gsilano Here you are
[image]

gsilano, 2023-05-14:
I've found an explanation for the error you encountered along with a possible [solution](https://stackoverflow.com/questions/35856969/chrono-literals-is-not-a-namespace-name). However, I'm a bit confused because when I look at your screen, and in the `rotors_simulator` repository (`dev\noetic` branch) I can't locate the exact source of this error. It seems there is no folder named `include\ignition` within it. Interestingly, the code compiles without any issues on my machine. This leads me to wonder if there are other files within your `catkin_ws folder` or your working space that might be causing conflicts.

gsilano, 2023-11-28:
Closing for inactivity

closed 2023-11-28 by gsilano