gsilano/BebopS #49: gzserver error: Look up (when running PX4 mavros example)

Bhardwaj-Harsh, 2023-07-10:
Running gazebo 11, ROS Noetic, ubuntu 20.04 LTS

following this: https://docs.px4.io/main/en/ros/mavros_offboard_python.html

Error when running the offboard_node example

C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://harsh-Inspiron-15-7510:45921/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
[... 266 lines cut ...]
[ INFO] [1688984190.742620434]: Plugin vision_speed_estimate loaded
[ INFO] [1688984190.744420290]: Plugin vision_speed_estimate initialized
[ INFO] [1688984190.744575411]: Plugin waypoint loaded
[ INFO] [1688984190.747256294]: Plugin waypoint initialized
[ INFO] [1688984190.747300168]: Plugin wheel_odometry blacklisted
[ INFO] [1688984190.747452075]: Plugin wind_estimation loaded
[ INFO] [1688984190.747769294]: Plugin wind_estimation initialized
[ INFO] [1688984190.747890556]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1688984190.747916118]: Built-in MAVLink package version: 2023.6.6
[ INFO] [1688984190.747943956]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[ INFO] [1688984190.747969016]: MAVROS started. MY ID 1.240, TARGET ID 1.1
gzserver: symbol lookup error: gzserver: undefined symbol: _ZN6gazebo4util9LogRecord8InstanceEv
[gazebo-3] process has died [pid 103418, exit code 127, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/harsh/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/worlds/empty.world __name:=gazebo __log:=/home/harsh/.ros/log/d62eebdc-1f0a-11ee-8b81-7133e4e5b070/gazebo-3.log].
log file: /home/harsh/.ros/log/d62eebdc-1f0a-11ee-8b81-7133e4e5b070/gazebo-3*.log
[INFO] [1688984190.863862, 0.000000]: Loading model XML from file /home/harsh/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/iris/iris.sdf
[INFO] [1688984190.866592, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1688984191.032573478]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1688984191.033757908]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
[gazebo_gui-4] process has died [pid 103430, exit code 134, cmd /opt/ros/noetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/harsh/.ros/log/d62eebdc-1f0a-11ee-8b81-7133e4e5b070/gazebo_gui-4.log].
log file: /home/harsh/.ros/log/d62eebdc-1f0a-11ee-8b81-7133e4e5b070/gazebo_gui-4*.log

Please help!

gsilano, 2023-07-10:
Could you be more specific? As it is, it's very difficult to figure out what's wrong.

gsilano, 2023-11-28:
Closing for inactivity

closed 2023-11-28 by gsilano