gsilano/BebopS #50: drone simulation to live drone flight

Dk-0, 2023-11-24:
The reality of bebop2 drone flight requires publishing the cmd_vel topic to the drone, and each piece of information under the topic is: linear.x and linear.y sections set the pitch and roll angles of the Bebop in degrees, respectively; linear.z controls the vertical speed of the Bebop in metres per second; angular.z controls the z controls the Bebop's vertical speed in metres per second; angular.z controls the Bebop's rotational speed (around the Z axis) in degrees per second. The details are shown in the figure below:
[image]

And there is no published topic for this in UAV simulation. All topics are shown below:
[image]

How should I control the real bebop2 drone to fly according to the simulation effect?

gsilano, 2023-11-24:
Hi! Have you checked the "dev/bebopInterface" and "dev/sphinx" branches?

gsilano, 2023-11-28:
Hello! Sorry for the delay. As I said in my previous comment, you should jump into the other branches. There, you will find scripts and interfaces that you are looking for. Sorry if I'm not helping you any further, but this is a very intensive period for me. I hope this helps.

Dk-0, 2023-11-29:
Thank you so much for taking the time to respond to my issue. Your help has been greatly appreciated! Wishing you all the best with your research! Take care!

closed 2023-11-29 by gsilano